realsenseai/librealsense #1281: Synchronize to 60hz

maciekd, 2018-03-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.09.00 | 
| Operating System & Version | Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.4.0-116-generic | 
| Platform                                 | Intel NUC |
| SDK Version                            |2.10.0  | 

### Issue Description
We've been synchronizing our entire system to the frame rate from the RealSense cameras. With the new RS400 cameras the frame rate drops to 30fps sometimes (the cameras autoexposure is trying to compensate for low light conditions).

Is there any other way to synchronize to the cameras and get a fixed 60 frame rate, independent of the auto exposure compensation?

pavolbauer, 2018-03-13:
Hi, 
I would be also interested in this, since we observed huge fluctuations of the FPS (in a range from 90fps to 20fps although the camera was set to 90fps) in a quite static environment. 
Is the only way to go to disable the autoexposure? In that case the maximal exposure at 90 ftps would be 11ms, would that equal setting the "depth exposure" to 1100 units?
Cheers
Pavol

agrunnet, 2018-03-13:
11000 is equivalent to 11ms.

RealSense-Customer-Engineering, 2018-03-19:
[Realsense Customer Engineering Team Comment] 
@maciekd  What camera are you using? And what's the resolution when you got the issue?

I tried D410 in my hand with v2.10.0 and FW 5.8.9 but couldn't reproduce the issue when AE enabled and move the camera from good light to low light.

maciekd, 2018-03-19:
Hi,

I believe it was a 415 with firmware 05.08.09.00. 

We've switched to running only 435 now and we set RS2_OPTION_AUTO_EXPOSURE_PRIORITY to zero. Since then the camera seems to be running at 60fps all the time.

RealSense-Customer-Engineering, 2018-03-26:
[Realsense Customer Engineering Team Comment] 
@maciekd  You can try D415 with the latest firmware v5.9.2 to see if you still have issue.

RealSense-Customer-Engineering, 2018-04-10:
[Realsense Customer Engineering Team Comment] 
@maciekd How about the result of FW 5.9.2? Is it ok to close this issue?

maciekd, 2018-04-11:
We're swamped with work chasing a deadline and we've gone over to using only d435. Unfortunately I haven't had time to test this again. But, with the setup we have now everything is working fine.

RealSense-Customer-Engineering, 2018-04-11:
[Realsense Customer Engineering Team Comment] 
@maciekd Glad to hear that everything is fine. Will close this ticket accordingly. Thanks!

closed 2018-04-16 by RealSense-Customer-Engineering