realsenseai/librealsense #1282: Question: select one device, but open another one?

iamyuiwong, 2018-03-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 and D435 | 
| Firmware Version                   | SR300 3.21.0.0  D435 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.13.0-36-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.1                        | 

### Issue Description

Hi, I have two cameras, SR300 and D435, SR300 cannot work now, I'll follow #1186 to fix,
D435 is OK.
And the question is, I connected two cameras, when I selected SR300 (I tried both by serial number and device name), but always opened D435, I don't know whether SR300 cannot work, but information I get from the device is about SR300, and the realsense-viewer is OK, so what should I do to solve this problem? 
Thanks.

zivsha, 2018-03-05:
Hi @iamyuiwong , are you using the pipeline to start streaming?
If so,please note that you should pass a `rs2::config` instance to the `pipeline.start(cfg)` method. For example:
```cpp
rs2::config cfg;
rs2::pipeline pipe;
cfg.enable_device(sr300_serial);
pipe.strat(cfg);
```
This will only start streaming with the specified sr300 camera. If you call `pipe.strat()` (without passing the config object, then the default configuration will be chosen and D435 will start streaming if it is connected.

If you are not using the pipeline, please post the code you use and we'll see what you are missing.

iamyuiwong, 2018-03-05:
Hi @zivsha, yes, I used `pipeline.start(cfg)` to start streaming, and I've found where wrong ?:
I query a device list, then match SN or name, then I save the device, but finally when I start streaming(some streams with specific width, height ... I set to then `cfg`), then
I just create a default pipeline then start: `rs2::pipeline pipeline; pipeline.start(cfg);`,
not contact to the device I selected? So, I want to know how to start a specific device(I selected)
or other way to select the device?

Oh, I'm writing a ROS camera driver from the realsense ros sample and used SDK 2.0 ... almost done but now not very clear how to handle this.

zivsha, 2018-03-05:
The way to do it is like I wrote in the code above using: 
https://github.com/IntelRealSense/librealsense/blob/5121ab030c939632d4d4a36ac999a937c3b2e170/include/librealsense2/hpp/rs_pipeline.hpp#L201-L209

iamyuiwong, 2018-03-05:
Oh, I got it! many thanks!

zivsha, 2018-03-06:
No problem, can we close the issue?

iamyuiwong, 2018-03-13:
Yes, thanks.

closed 2018-03-15 by RealSense-Customer-Engineering