realsenseai/librealsense #1283: D415 Device Recognized on Power Up, Not on Reboot

jallard-sr, 2018-03-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16.04.4 LTS  | 
| Kernel Version (Linux Only)    |  4.4.0-116-generic #140-Ubuntu SMP Mon Feb 12 21:23:04 UTC 2018  | 
| Platform                                 | NUC with dual Intel(R) Core(TM) i5-6260U CPU @ 1.80GHz cores |
| SDK Version                            |  librealsense tag 2.10.0  | 

### Issue Description
When rebooting this type of NUC computer with a D415 plugged in, the D415 does not register properly as a USB device, with the following error in dmesg output. However, when it is power cycled, the D415 registers fine.

`[    2.477775] usb 2-4: device not accepting address 2, error -71`

I've attached the USB relevant bits of `dmesg` and the output of `lsusb -t` from a good start on power up, and from a bad start on reboot. Oddly enough, this does not occur with newer NUC models with dual Intel(R) Core(TM) i5-7260U CPU @ 2.20GHz cores, where the device registers fine on reboots.

[dmesg-bad.txt](https://github.com/IntelRealSense/librealsense/files/1781754/dmesg-bad.txt)
[dmesg-good.txt](https://github.com/IntelRealSense/librealsense/files/1781755/dmesg-good.txt)
[lsusb-t-bad.txt](https://github.com/IntelRealSense/librealsense/files/1781756/lsusb-t-bad.txt)
[lsusb-t-good.txt](https://github.com/IntelRealSense/librealsense/files/1781757/lsusb-t-good.txt)

jallard-sr, 2018-03-05:
Looks like dmesg output on failure didn't upload correctly. Here it is again.

[dmesg-bad.txt](https://github.com/IntelRealSense/librealsense/files/1782310/dmesg-bad.txt)

RealSense-Customer-Engineering, 2018-03-13:
[Realsense Customer Engineering Team Comment] 
@jallard-sr, in the BIOS settings try setting the xHCI mode to Enabled (not Auto or Smart Auto) under Devices/USB.

ioreed, 2018-03-20:
This setting was already enabled in the BIOS as described: xHCI mode = Enabled

RealSense-Customer-Engineering, 2018-03-21:
[Realsense Customer Engineering Team Comment] 
Although I haven't seen this with the 415 it is similar to an issue with the 435 - except xHCl mode = enabled was a usable work-around (newer NUCs only have xHCI). For the f/w 5.9.2.0 fixed the problem for the 435 cameras so you can try that if you wish. https://downloadcenter.intel.com/download/27522/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t

RealSense-Customer-Engineering, 2018-04-05:
[Realsense Customer Engineering Team Comment] 
@jallard-sr did you try 5.9.2.0 f/w and/or do you still see the issue?

tedtman, 2018-04-25:
@jallard-sr I assume that you are no longer having this issue and I will close this ticket. If you still have this or any other problem please feel free to open another issue.

Thanks.
Ted.
Intel RealSense Support.

closed 2018-07-26 by dorodnic