realsenseai/librealsense #1286: Software trigger and saving to CSV

pavolbauer, 2018-03-06:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.0.02 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.0                     | 
----------------------------------------------------------------------------------------------------

Hi,

I have two questions about the C++ RealSense SDK 2.0 in combination with the D435 cam.

1) Is there a way to trigger the acquisition of a frame from within the code? So, instead of defining a stream and using wait_for_frames() using a get_frame() function that delivers me the next frame whenever the camera is ready to fire the next time? Optimally this should also respect the frame rate defined in the settings.

2) What is the best and memory-friendliest way to store the captured depth data as uint16 in a large CSV file? The current SDK provides stbi calls for exporting as png, etc. but I would just like to have the raw data being appended onto a large CSV or TAB file that I can postprocess later.

Thanks, Best
Pavol

zivsha, 2018-03-06:
Hi Pavol,
1. Yes, either use `if(pipeline.poll_for_frame(&frameset))`, or `sensor.start(on_frame)`. The first will get a framset if it is available, the second will callback when a frame is available.
2.I don't think there is a memory friendly way of doing it since it will always use the same amount of memory. We don't provide anything like that but it should be pretty straight forward to write csv files with raw data. Let me know if you need any help with that

pavolbauer, 2018-03-06:
Hi Zivsha,

Thanks for your prompt and precise answer!
When I wrote memory-friendly I didnt mean it with respect to the output file size, but rather with how the buffers are used, so this can be done efficiently at high frame rates.
When I do it the 'naive' way and loop over the depth frame pixel by pixel in order to pass this to an output steam this is notoriously slow and cant keep up with 90 fps, for example.

Best
Pavol

zivsha, 2018-03-06:
If you want to have the most control, and lowest latency I would go with the callback and copy the data to another thread to do the IO. This question is more of a software design one, so I don't want to make the choice for you. What you should know is that copying a `rs2::frame` does not deep copy the image, rather only adds a reference count to it, so pushing it to a frame queue is cheap, but is limited by the `RS2_OPTION_FRAMES_QUEUE_SIZE` (the sensor will stop providing frames if you keep more than the max queue size).  You can either copy the data (with `memcpy` or similar) to your own held memory or use the `frame.keep()` method (which has the caveat of possibly allocating memory per frame instead of a closed buffer pool- which in turn could impact performance).
As jigsaw said, "...make your choice".

pavolbauer, 2018-03-06:
Hi Zivsha,
Thanks for the great advise.
I will go with memcpy and and try to do it in another thread, as you suggested. 
Best
Pavol

closed 2018-03-06 by pavolbauer

kyranf, 2018-06-08:
@pavolbauer  if you are only doing this testing for a reasonably short duration (i.e minutes, or tens of seconds) then you can simply copy each new frame into an array or push them into a vector of rs2::frame objects and when the processing is signalled as done (maybe on a control+C event, or other trigger to stop and save) go through the list and process the frames and dump them into the CSV without worrying about realtime performance.