realsenseai/librealsense #1287: SDK error when creating pipeline

GuidoBartoli, 2018-03-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16.04.3  | 
| Kernel Version (Linux Only)    |  4.13.0-36-generic                                         | 
| Platform                                 | Compulab FitPC  |
| SDK Version                            |  2.10.1                          | 

I'm writing a ROS node for embedding Intel RealSense **D435** perception into a robotics project.
I installed the SDK 2 following the [official guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) (I encountered some problems with `sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE` because GPG key could not be retrieved, but I managed to manually download it and add to keyring).

Aside from some glitches (some warning and error popups), _RealSense Viewer_ is working properly now.

After that, I wrote a very simple C++ test node to try fetching frames following the "rs-capture.cpp" example provided (I already updated my `CMakeLists.txt` to correctly compile this node):

```
#include <ros/ros.h>
#include <librealsense2/rs.hpp>

int main(int argc, char** argv)`
{
    // ROS initialization
    ros::init(argc, argv, "realsense_driver");
    ros::NodeHandle nh;
    ros::NodeHandle np("~");

    // Device initialization
    rs2::pipeline pipe;
    pipe.start();
    ROS_INFO("[REALSENSE] Streaming pipeline started");
    while (ros::ok())
    {
        rs2::frameset frames = pipe.wait_for_frames();
        rs2::depth_frame depth = frames.get_depth_frame();
    }

    return 0;
}
```

After issuing the usual `catkin_make` command I received many error messages (I don't report them here for brevity), so I reduced the body to just `rs2::pipeline pipe;` to better identify the problem and that's what I receive:

```
CMakeFiles/realsense_driver.dir/src/RealSenseDriverNode.cpp.o: In function `rs2::error::error(rs2_error*)':
/usr/include/librealsense2/hpp/rs_types.hpp:76: undefined reference to `rs2_get_error_message'
/usr/include/librealsense2/hpp/rs_types.hpp:78: undefined reference to `rs2_get_failed_function'
/usr/include/librealsense2/hpp/rs_types.hpp:78: undefined reference to `rs2_get_failed_function'
/usr/include/librealsense2/hpp/rs_types.hpp:79: undefined reference to `rs2_get_failed_args'
/usr/include/librealsense2/hpp/rs_types.hpp:79: undefined reference to `rs2_get_failed_args'
/usr/include/librealsense2/hpp/rs_types.hpp:80: undefined reference to `rs2_get_librealsense_exception_type'
/usr/include/librealsense2/hpp/rs_types.hpp:81: undefined reference to `rs2_free_error'
CMakeFiles/realsense_driver.dir/src/RealSenseDriverNode.cpp.o: In function `rs2::error::handle(rs2_error*)':
/usr/include/librealsense2/hpp/rs_types.hpp:127: undefined reference to `rs2_get_librealsense_exception_type'
CMakeFiles/realsense_driver.dir/src/RealSenseDriverNode.cpp.o: In function `rs2::context::context()':
/usr/include/librealsense2/hpp/rs_context.hpp:84: undefined reference to `rs2_create_context'
/usr/include/librealsense2/hpp/rs_context.hpp:84: undefined reference to `rs2_delete_context'
CMakeFiles/realsense_driver.dir/src/RealSenseDriverNode.cpp.o: In function `rs2::pipeline::pipeline(rs2::context)':
/usr/include/librealsense2/hpp/rs_pipeline.hpp:349: undefined reference to `rs2_create_pipeline'
/usr/include/librealsense2/hpp/rs_pipeline.hpp:349: undefined reference to `rs2_delete_pipeline'
collect2: error: ld returned 1 exit status
```
It seems a problem inside the SDK, but I really cannot understand why it cannot find its own structures and types... does anyone knows what I'm missing here?

Thanks,
Best regards

zivsha, 2018-03-06:
Hi @GuidoBartoli ,
These look like regular linkage issues, telling you that the linker can't find the implementation of these functions (`rs2_*`). It looks like you forgot to tell `CMake` to link the `realsense2` library.

Please make sure that you properly installed librealsense, make sure that `/usr/lib` or `/usr/local/lib` has `realsense2.so`. You might want to try to run `ldconfig` to tell the linker to refresh itself.

On another note- we provide a [ROS wrapper](https://github.com/intel-ros/realsense/releases), you might want to have a look at it.

GuidoBartoli, 2018-03-06:
Wow, what a blunder from me... I forgot to link the node to library!
```
add_executable(
    realsense_driver
    src/RealSenseDriverNode.cpp
)
add_dependencies(realsense_driver ${catkin_EXPORTED_TARGETS})
target_link_libraries(
    realsense_driver
    realsense2
    ${catkin_LIBRARIES}
)
```
Thanks for pointing it out! :)

I will take a look to the ROS wrapper, I did not know it existed.

closed 2018-03-06 by GuidoBartoli

dmitryshendryk, 2018-04-22:
@GuidoBartoli Hello, i have the same error in Eclipse. I've binded all libraries. But still has it problem. Did you update your Makefile?

GuidoBartoli, 2018-05-11:
@dmitryshendryk Yes, I edited the `CMakeLists.txt` file as I reported in my last post. The node executable was missing the linking to the **realsense2** library.

karry3775, 2020-01-15:
@zivsha  I wrote a simple cpp file to check the working of librealsense2.

#include <librealsense2/rs.hpp>

int main(int argc, char * argv[]){
  window app(1280, 960, "CPP MUlti-Camera Example");
  rs2::context ctx;
  return 0;
}

I keep getting the undefined reference error for rs2. I run the following command
 g++ to_check_librealsense2.cpp -lGL -lGLU -lglut
where the linkers -lGL, -lGLU and -lglut are for openGL. Before adding these linkers I was getting undefined reference for even openGL functions. Is there a similar linker function for rs2. Thanks!