realsenseai/librealsense #1289: C# point cloud

michalstepien, 2018-03-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.9.1 }                          | 

### Issue Description
Hello, is it possible to get point cloud from C# wrapper?

zivsha, 2018-03-06:
Yes, but not with 2.9.1, it was added in 2.10.1 I think.
We'll add texture mapping to pointcloud as well this week.

zivsha, 2018-03-07:
You can see example usage of the `PointCloud` object in the Unity project:
https://github.com/IntelRealSense/librealsense/blob/d952352f719b6ab723e7f1139a835e4300387f1f/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/PointCloudGenerator.cs#L63-L66

Zeraphil, 2018-03-07:
Hi zivsha. Also interested in this. Where is the PointCloud object defined? It's not in the CSharp or Unity wrapper folders...

zivsha, 2018-03-08:
Hi, apparently this is only in development branch and was yet to merge to master.
You can find it here:
https://github.com/IntelRealSense/librealsense/blob/4b311e22bacd2ef10e30f2e9522419e1216e1147/wrappers/csharp/Intel.RealSense/Processing.cs#L170

dorodnic, 2018-03-18:
Should be available in master right now. @michalstepien , @Zeraphil feel free to open new issues if you have follow-up questions

closed 2018-03-18 by dorodnic