realsenseai/librealsense #1290: .bag file to Matlab or video file

marcalessio, 2018-03-06:
Hello,

I recorded a depth video with the  Intel.RealSense.Viewer.exe saving a .bag file.

How can I load the video I recorded within Matlab to analyze it and perform some computations? Is there a way to convert it in a video format?

Regards,
Marco.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)                                        | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          |

zivsha, 2018-03-06:
Hi @marcalessio .bag files are saved as ROSbag format, and can be used with common ros tools, I'm not sure if there are any that provide you with a way to open them in Matlab.There are C++ and Python APIs to read bag files if that helps.

We are working on a Matlab wrapper for the SDK so when it's ready you will be able to load the files with it but it's work in progress (see PR #1038).

apoorva2398, 2018-07-13:
You can use MATLAB's [Robotics Systems Toolbox](https://www.mathworks.com/products/robotics.html) to read each frame as follows:

`bag = rosbag('file.bag');  % Load the rosbag.`

`rosbag info 'file.bag';` 
`% Displays information in the command window. Use bagInfo = rosbag('info', 'file.bag') to get the information as a structure in a script.`

`bSel = select(bag, 'Topic', '/device_0/sensor_0/Infrared_1/image/data');   % Select a specific topic from the above information. It should end with /image/data if you want the frames.`

`msgs = readMessages(bSel); % Read all the messages (here, frames) in that topic as an nx1 cell array.`

`% Iterate through the cell array and read each frame.`
`for i = 1 : numel(msgs)`
`img = readImage(msgs{i});`
`% Do something with the frame (img).`
`end`

RealSense-Customer-Engineering, 2018-07-24:
[Realsense Customer Engineering Team Comment]
thanks for your share @apoorva2398,
hi @marcalessio, wish this works for you.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @marcalessio,
do you need more support on this topic?

closed 2018-08-20 by RealSense-Customer-Engineering

taubar, 2019-07-02:
Hi, i'm experiencing similar problems (visualising data) using D435 on MacOS.
After running the bSel, i get 

bSel = 
  BagSelection with properties:

           FilePath: '/Volumes/RonyPaz_08-9346932/20190408_140444.bag'
          StartTime: 0.0365
            EndTime: 1.2135e+03
        NumMessages: 18219
    AvailableTopics: [1×3 table]
    AvailableFrames: {0×1 cell}
        MessageList: [18219×4 table]   

so the msgs = readMessages(bSel); % Read all the messages (here, frames) in that topic as an nx1 cell array.

returns 

Error using robotics.ros.BagSelection/deserializeMessages (line 641)
Java exception occurred:
java.lang.IndexOutOfBoundsException: Invalid combined index of 1229408318, maximum is 34538290
	at org.jboss.netty.buffer.SlicedChannelBuffer.<init>(SlicedChannelBuffer.java:46)
	at org.jboss.netty.buffer.HeapChannelBuffer.slice(HeapChannelBuffer.java:200)
	at org.jboss.netty.buffer.AbstractChannelBuffer.readSlice(AbstractChannelBuffer.java:323)
	at org.ros.internal.message.field.PrimitiveFieldType$14.deserialize(PrimitiveFieldType.java:599)
	at org.ros.internal.message.field.PrimitiveFieldType$14.deserialize(PrimitiveFieldType.java:565)
	at org.ros.internal.message.field.ValueField.deserialize(ValueField.java:68)
	at org.ros.internal.message.DefaultMessageDeserializer.deserialize(DefaultMessageDeserializer.java:45)
	at org.ros.internal.message.field.MessageFieldType.deserialize(MessageFieldType.java:111)
	at org.ros.internal.message.field.MessageFieldType.deserialize(MessageFieldType.java:38)
	at org.ros.internal.message.field.ValueField.deserialize(ValueField.java:68)
	at org.ros.internal.message.DefaultMessageDeserializer.deserialize(DefaultMessageDeserializer.java:45)

Error in robotics.ros.BagSelection/readMessages (line 278)
                msgs = obj.deserializeMessages(obj.MessageList, rows);

would appreciate any help. :)

lramati, 2019-08-20:
Please try to avoid reviving threads that are no longer relevant. In the past year, we have released a [Matlab wrapper for realsense](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab) that you might want to look into.
The specific error you've posted above seems to be a problem with Matlab's Robotics Systems Toolkit.