realsenseai/librealsense #1291: R200 stability and UVCvideo errors

doisyg, 2018-03-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200| 
| Firmware Version                   | 1.0.72.06 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0-36-generic                         | 
| Platform                                 | Laptop Asus and  multiple NUC6i5SYH  |
| SDK Version                            |  legacy                     | 

### Issue Description
I am trying to trace the origin of some instability i get on the R200 ROS driver (crash of the driver about once every 24h to once every week) and I wonder if the uvcvideo errors I get are related (that's why I post here) or if I can ignore them completely. Here is the relevant output from `dmesg`, the "failed to query" errors appear after I (successfully) start the ROS driver.

> 
> [118003.791133] usb 2-6: new SuperSpeed USB device number 2 using xhci_hcd
> [118003.813929] usb 2-6: New USB device found, idVendor=8086, idProduct=0a80
> [118003.813935] usb 2-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [118003.813940] usb 2-6: Product: Intel RealSense 3D Camera R200
> [118003.813943] usb 2-6: Manufacturer: Intel Corp
> [118003.813947] usb 2-6: SerialNumber: SN_2401000982
> [118003.816441] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [118003.819466] uvcvideo 2-6:1.0: Entity type for entity Extension 2 was not initialized!
> [118003.819474] uvcvideo 2-6:1.0: Entity type for entity Camera 1 was not initialized!
> [118003.820516] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [118003.823275] uvcvideo: Unable to create debugfs 2-2 directory.
> [118003.823481] uvcvideo 2-6:1.2: Entity type for entity Camera 4 was not initialized!
> [118003.823954] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.2/input/input21
> [118003.825244] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [118003.828074] uvcvideo: Unable to create debugfs 2-2 directory.
> [118003.828255] uvcvideo 2-6:1.4: Entity type for entity Processing 7 was not initialized!
> [118003.828260] uvcvideo 2-6:1.4: Entity type for entity Camera 6 was not initialized!
> [118003.828498] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.4/input/input22
> [118035.952858] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.003651] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.054114] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.104678] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.155225] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.205681] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.256206] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.307194] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.357670] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.408077] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.458546] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.508972] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.559416] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
> [118036.609853] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
>

ev-mp, 2018-03-06:
Hello @doisyg ,
It is not clear what do you mean by `crash of uvc driver` - can you provide kernel/application logs taken during this event ?

According to the timestamps the reports are fired 32 seconds after connecting R200 to Host or Boot-up. I.e. it occurred immediately upon application start. Connecting it to the crash that occurs once in 24 hours seem unreasonable.

As to the error itself - it is one of standard UVC controls (e.g Gain,Exposure) that could not be applied in a timely manner.
Linux UVC driver does not support async controls so Librealsense would retry the same command in loop till a proper response is received/timeout. (you may notice 50ms periods between retries). This situation is not standard yet quite common, and does not indicate functionality issues.

doisyg, 2018-03-07:
Hi @ev-mp ,
Thanks for your prompt answer. I meant crash of the ROS driver: https://github.com/intel-ros/realsense/tree/1.8.1
I will post the logs once I will have them, but it is good to now that I can discard these UVC errors.

RealSense-Customer-Engineering, 2018-04-11:
[Realsense Customer Engineering Team Comment] 
Hi @doisyg,

Still need any help? I didn't see this issue on my test configuration: Ubuntu 16.04 + kernel 4.14.15 + ROS wrapper build 1.8.1 + librealsense 1.12.1.
Please refer to http://wiki.ros.org/realsense_camera to have the correct installation. After librealsense package installed, only need to install the prebuilt ROS Debian packages.
$ sudo apt-get install 'ros-*-realsense-camera'

Then, you should be able to run below:
$ roslaunch realsense_camera r200_nodelet_default.launch

doisyg, 2018-04-11:
Thanks for your answer, I still have stability issues (1 or 2 crash a week, but we can detect it and relaunch immediately) but i guess it is more related to usb power management, maybe triggered by a Ubuntu system upgrade. It appears on multiple NUC with different r200 several month ago after working flawlessly for 1 year and a half.

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Appears to be a platform issue

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
hi @doisyg,

since it was working flawlessly for 1.5 years, i think this more like platform issue, like you mentioned, usb power management issue.

closed 2018-07-23 by RealSense-Customer-Engineering