realsenseai/librealsense #1292: Compilation error with -Wall in clang (patch attached)

richardroberts, 2018-03-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.9.0                          | 

### Issue Description
Minor compilation error with -Wall (which we're required to use here), patch attached here: 
[compilation_error_fix_richardroberts.patch.txt](https://github.com/IntelRealSense/librealsense/files/1786506/compilation_error_fix_richardroberts.patch.txt)

By the way, I'd be glad to submit these minor fixes as pull requests if that makes things easier!  Is there a permission change that I need to be able to push to a new branch?  Currently I get this error:
`ERROR: Permission to IntelRealSense/librealsense.git denied to richardroberts.
fatal: Could not read from remote repository.`

ev-mp, 2018-03-06:
Hi @richardroberts , thanks for the patch.
You are welcomed to create Pull Request and submit this patch to the `development` branch for review and merge.

zivsha, 2018-03-06:
@richardroberts yo can fork (upper right corner of GitHub) the repo and push to a branch there, then submit a PR to development on librealsense

RealSense-Customer-Engineering, 2018-03-27:
[Realsense Customer Engineering Team Comment] 
Hi, @richardrobert,
Didn't see your PR, have you created PR?

richardroberts, 2018-04-12:
Thanks for the help Sergey, in the future I'll try again to create a pull request.

closed 2018-04-15 by dorodnic