realsenseai/librealsense #1294: Support of hardware timestamps

TheMikeyR, 2018-03-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04.4)  | 
| Kernel Version (Linux Only)    |  4.14.13-041413-generic   / 4.4.38-tegra                                  | 
| Platform                                 | PC / Nvidia Jetson  |
| SDK Version                            |  v2.10.0                       | 

### Issue Description
Currently I only have System Time available from the `realsense-viewer`, it might be due to wrong kernel on pc, but the Jetson have a correct kernel version (though tegra version). Can you tell about plans on supporting more kernels / platforms? So I'm able to know which devices I'm able to use as a host to retrieve hardware timestamps for more reliable sync purposes.

The funny thing is that when running `rs-save-to-disk.cpp` I get a timestamp out for the color camera but not the depth stream, in the `realsense-viewer` it shows system time when clicking on info when showing the color stream. I assume backend timestamp is hardware timestamp or am I wrong? 

I don't have any issues when following the installation guide for linux, no errors and all libraries installed.

ev-mp, 2018-03-07:
Hi @TheMikeyR , 
In order to get HW timestamp on Linux you need to apply [metadata kernel](https://github.com/IntelRealSense/librealsense/blob/development/scripts/realsense-metadata-ubuntu-xenial-master.patch) patch, in addition to the Depth Formats patch to Tegra kernel as part of the [Video4Linux backend preparations](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md#prerequisites).

You can see how it's done for Ubuntu here : https://github.com/IntelRealSense/librealsense/blob/development/scripts/patch-realsense-ubuntu-xenial.sh#L71

TheMikeyR, 2018-03-08:
@ev-mp so it should work out of box with my Ubuntu laptop right? The weird thing is that I still see "System Time" when looking through `realsense-viewer` and I've tried to run the [script](https://github.com/IntelRealSense/librealsense/blob/development/scripts/patch-realsense-ubuntu-xenial.sh) several times, and also followed the [Video4Linux backend preparations](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md#prerequisites) Is it due to my kernel is not supported? `4.14.13-041413-generic`

ev-mp, 2018-03-08:
Hi @TheMikeyR , We provide kernel patches for Ubuntu LTS kernels, currently 4.4, 4.10 and 4.13 as part of the manual installation guide or DKMS.
The patches were not verified with  4.14, as according to [Ubuntu sources](https://wiki.ubuntu.com/Kernel/LTSEnablementStack) this version is not going to receive LTS stamp.

Nevertheless, porting from 4.13 to 4.14 should be relatively straight-forward.
Let us know whether you encounter specific issues and we will try to assist.

ev-mp, 2018-03-25:
Hello @TheMikeyR , I'm closing it as answered.
Feel free to reopen if you have issues with patching LTS kernels.

closed 2018-03-25 by ev-mp