realsenseai/librealsense #1297: get_infrared_frame python wrapper

TheMikeyR, 2018-03-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04.4)  | 
| Kernel Version (Linux Only)    |  4.14.13-041413-generic                                 | 
| Platform                                 | PC  |
| SDK Version                            |  v2.10.0                       | 
| Language                            |  Python                       | 

### Issue Description
Any plan on adding `get_infrared_frame` to the python wrapper or there any workaround to get both infrared streams? Currently I'm able to get one of the infrared stream using the `frameset.first(rs.stream.infrared)` call. 

I suspected that it should be available in the python wrapper but can't seem to find it anywhere https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/python.cpp#L419

Currently you are able to call `get_infrared_frame(index)` in cpp according to https://github.com/IntelRealSense/librealsense/issues/1140#issuecomment-364816438 
 I can see the function is defined here https://github.com/IntelRealSense/librealsense/blob/9437bc24c0f4ce16193f1045ceea703a8151b96d/include/librealsense2/hpp/rs_frame.hpp#L691

It would be nice with the same functionality to python.

zivsha, 2018-03-07:
Hi @TheMikeyR ,
Yes we can add it, in fact you can also add it :)
Currently what you can do is simply iterate over the framset:
```py
for frame in pipe.wait_for_frames():
    print frame.profile
```

TheMikeyR, 2018-03-07:
@zivsha thanks, i've just realised when recording data from realsense-viewer that it only record one infrared stream even though both is activated. Not sure if that is intended or bug? I've tried to enable both infrared and no depth, record, load bag file and post index like above and it only shows index 2. 

If I enable both streams from camera and use the above, I get both index 1 and 2. 

I might add get_infrared_stream if I can figure out how, I'm not the strongest in cpp yet.

zivsha, 2018-03-07:
>  i've just realised when recording data from realsense-viewer that it only record one infrared stream even though both is activated. Not sure if that is intended or bug?

Can you please elaborate on this? The viewer should (and as far as I ca see, does) record both infrared streams. This works for me with or without depth stream. Can you please make sure that if you open the file with the viewer, all streams are available?

Regarding infrared index - The index might also be 0 sometimes, depending on which streams are active. If you run `rs-enumarate-devices` you can see that some stream profiles have index 0 (no index will appear) and some have 1 or 2. 

Regarding adding the function to python, `pybind11` did a truly amazing job to make it simple to bind cpp code to python, so to add binding to that function you just have to add the following line 
```cpp
        .def("get_infrared_frame", &rs2::frameset::get_infrared_frame, "index"_a = 0)
```
next to:
https://github.com/IntelRealSense/librealsense/blob/5121ab030c939632d4d4a36ac999a937c3b2e170/wrappers/python/python.cpp#L419

TheMikeyR, 2018-03-08:
@zivsha oh yeah my bad, they are both being recorded (when I open it in the viewer all streams are available).

That seems extremely easy! I will try it out, thanks :+1:

TheMikeyR, 2018-03-08:
It worked wonders, thanks @zivsha I made a PR to get it added. For others who are coming across this, here is an example on how to get each IR stream.
```
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 1280, 720, rs.format.y8, 30)
config.enable_stream(rs.stream.infrared, 2, 1280, 720, rs.format.y8, 30)
pipeline.start(config)
frames = pipeline.wait_for_frames()
ir1_frame = frames.get_infrared_frame(1) # Left IR Camera, it allows 0, 1 or no input
ir2_frame = frames.get_infrared_frame(2) # Right IR camera
```

closed 2018-03-08 by TheMikeyR

zivsha, 2018-03-08:
Awesome, glad to hear it all works.

Woodstock94, 2018-12-12:
I just solved my issue by substituting 
`image = np.asanyarray(ir1_frame)`
with
`image = np.asanyarray(ir1_frame.get_frame())`

pedrombmachado, 2019-02-21:
@Woodstock94 you mean:
`image = np.asanyarray(ir1_frame.get_data())`
because there aren't any get_frame() method. :p 

```
import pyrealsense2 as rs
import numpy as np
import cv2

points = rs.points()
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
profile = pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        ir1_frame = frames.get_infrared_frame(1)
        if not ir1_frame:
            continue
        image = np.asanyarray(ir1_frame.get_data())
        cv2.namedWindow('IR Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('IR Example', image)
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()
```

Woodstock94, 2019-02-21:
yes, sure. I messed it up!^^

neocubist, 2019-05-30:
I believe this may be the solution I am looking for. I am a newbie to Python. How do I use the above code to extract the raw images?

neocubist, 2019-05-30:
Update: I have installed the necessary packages and managed to get the program running. This was able to stream the IR and also view the dots. However I was unable to capture an image. It would be great if someone could show me how to do that.
Also once the above program begins running, I cannot close the window to exit. Needed to break routine using ^c. Was there a fix for that?

pedrombmachado, 2019-05-30:
Hi @neocubist,
replace line `cv2.imshow('IR Example', image)` by `cv2.imwrite( "<filename>", image)` for saving the image.
replace `while True` by `while flag` and set the `flag=False` to stop. The answer to your next question "How do I change the flag?" is google it.
I hope it helps.
Bw
Pedro