realsenseai/librealsense #1298: how to install the firmware of a realsense D435 on Ubuntu

GustavZ, 2018-03-07:
how to install the firmware of a realsense D435 on Ubuntu 16.04?

realsense-viewer and rviz both crash or don't work properly with the realsense D435 so i thought the reason might be the missing firmware.

I downloaded it from (https://downloadcenter.intel.com/download/27522/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t) but i have no clue how to install it on ubuntu. executing the file does not work

TheMikeyR, 2018-03-07:
https://github.com/IntelRealSense/librealsense/issues/1268

GustavZ, 2018-03-07:
@TheMikeyR  Can you use the realsense without problems with rviz or the viewer?

For me rviz hangs and the viewer cant show anything but grey scale

TheMikeyR, 2018-03-07:
I don't use rviz
I don't have any issues regarding realsense-viewer

In the first run where I've only installed dependencies for the demo it showed weird stream, but finished the installation guide and then replugged the camera and it worked great with all the streams available. Did you finish the entire installation process @GustavZ ?

GustavZ, 2018-03-07:
ah replugging and a reboot did the job :)
Do you know how to capture the rgb stream with opencv?

btw i remember from another thread that you are also using the jetson for object detection.
I did some further work on it, reaching 30 fps+ on ssd mobilenet with visualization if you are interested @TheMikeyR

TheMikeyR, 2018-03-08:
@GustavZ so you did solve your question? 

Thanks a lot, I will look into it! :+1:

GustavZ, 2018-03-08:
yep the realsense is working thanks.

closed 2018-03-08 by GustavZ