realsenseai/librealsense #1299: Alternating IR emitter between on/off

sogartar, 2018-03-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.14.00 | 
| Operating System & Version |   Linux Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  development branch  | 

### Issue Description
I would like to be able to switch on/off the IR emitter rapidly between frames while the sensor is running. From the source code of librealsense it looks like it is not possible to know if an IR/Depth frame was acquired when the emitter was on or off. Do you know if turning on/off the emitter requires some heavy lifting on the device or the library? Could we also have this info in the frame metadata in a future release?

daB0bby, 2018-03-12:
This would be very interesting for my use case also

agrunnet, 2018-03-15:
This is a planned feature...Will come in a few months.

kirsitn, 2018-04-24:
This is very much of interest to me as well. I would like to capture every second frame with the emitter on and off, and at the moment it seems to take up to 6 frames before the emitter option is registered by the sensor. (Possibly even longer time, because I first set the emitter option and then capture a number of frames to see how long it takes for the emitter to turn on/off. I haven't tried setting the option inside the capture loop since it seems futile to do this as long as I don't know whether I'm actually capturing data with the right settings.)

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

This feature is enabled in firmware 5.11+ as you can see in the release note.

dorodnic, 2019-02-28:
To be more precise it is enabled starting firmware [5.11.4](https://downloadcenter.intel.com/download/28573/Latest-Firmware-for-Intel-RealSense-D400-Product-Family) for Global-Shutter devices only (D43x)

closed 2019-03-05 by RealSenseCustomerSupport

john-maidbot, 2023-07-21:
Hi, is this feature bad for the long term health of the emitter? Does Intel have any documentation on the number of cycles to expect from a realsense emitter? (e.g. on the d435)

MartyG-RealSense, 2023-07-22:
Hi @john-maidbot  There is not data available on the lifespan of the emitter.  The laser can wear out over time but as the emitter is a separate hardware component from the sensors then the camera can continue to operate, though with reduced quality of the depth image.

john-maidbot, 2023-07-22:
Thank you for the information.