realsenseai/librealsense #1300: uvcvideo kernel module is not loaded

HumamHelfawi, 2018-03-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  R200  | 
| Firmware Version                   | (Not able to get) | 
| Operating System & Version |   Ubuntu 16.04 LTS  | 
| Kernel Version (Linux Only)    |  (4.16.0-041600rc2-generic) | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy 1.12.1 } | 

I am having problem running my R200 on Ubuntu 16.04.

I am getting this error when I try to run any of the examples:

> RealSense error calling rs_create_context(api_version:11201):
>     uvcvideo kernel module is not loaded

As far as I understand, there is no need to run any patch since the Ubuntu kernel is 4.4+.

The same R200 with the same cable working on the same clone of PC that runs MS Windows.

ev-mp, 2018-03-08:
Hello @HumamHelfawi ,
What kernel notification do you receive when you plug in R200 device ?  (`dmesg -wT`)

Using experimental kernels may be challenging, and you need to verify that uvcvideo is enabled in the kernel configuration - try to run 
 `modinfo uvcvideo | grep version`   
to obtain driver's info.

If the uvc dirver is available but fails to load automatically,  you may try to force it by `sudo modprobe uvcvideo` (while monitoring the kernel log)

HumamHelfawi, 2018-03-08:
@ev-mp Thank you very much for taking time to write this reply. This is what I got when executing the commands you wrote :

> [Prş Mar  8 09:26:09 2018] usb 2-2: Product: Intel RealSense 3D Camera R200
> [Prş Mar  8 09:26:09 2018] usb 2-2: Manufacturer: Intel Corp
> [Prş Mar  8 09:26:09 2018] usb 2-2: SerialNumber: SN_2411009530
> [Prş Mar  8 09:26:10 2018] uvcvideo: version magic '4.4.98+ SMP mod_unload ' should be '4.16.0-041600rc2-generic SMP mod_unload '
> [Prş Mar  8 09:26:10 2018] uvcvideo: version magic '4.4.98+ SMP mod_unload ' should be '4.16.0-041600rc2-generic SMP mod_unload '
> [Prş Mar  8 09:26:10 2018] uvcvideo: version magic '4.4.98+ SMP mod_unload ' should be '4.16.0-041600rc2-generic SMP mod_unload '

`modinfo uvcvideo | grep version`

> version:        1.1.1
> srcversion:     D428578CC3112188068CD0D

`sudo modprobe uvcvideo`

> modprobe: ERROR: could not insert 'uvcvideo': Exec format error

ev-mp, 2018-03-09:
The logs indicate that the uvcvideo that was probed had been compiled with (at least the header files of ) kernel 4.4 and is not compatible with 4.16:  
>[Prş Mar 8 09:26:10 2018] uvcvideo: version magic '4.4.98+ SMP mod_unload ' should be '4.16.0-041600rc2-generic SMP mod_unload '

Is it possible that you tried to apply the kernel patches ? It also might be a leftover from previously installed kernel in case of upgrade).
The easiest way to rectify it probably by reinstalling 4.16RC kernel to revert the original version of the uvcvideo driver.

HumamHelfawi, 2018-03-09:
Thank you very much! Actually, yes I did run the scripts but I did that after I tried running and failed. However,I will search for a way to revert the original version of uvcvidio since reinstalling the kernel did not work for me.
Thank you again for your valuable help!

ev-mp, 2018-03-12:
Hello @HumamHelfawi , in most cases running
`sudo apt-get install --reinstall linux-image-generic linux-image`
should be sufficient to reset the modules.
Let us know whether you manage to stream depth with the original driver in place

HumamHelfawi, 2018-03-12:
Hi @ev-mp , After I run the instruction you wrote, I am not getting the old error anymore! thank you very much!

However, the system can not see my R200! when I plug it in and using dmesg -wT:

> [Mon Mar 12 16:17:55 2018] Hardware name: LENOVO 80SR/Toronto 5A1, BIOS 0XCN23WW 03/21/2016
> [Mon Mar 12 16:17:55 2018] Workqueue: usb_hub_wq hub_event
> [Mon Mar 12 16:17:55 2018] Call Trace:
> [Mon Mar 12 16:17:55 2018]  dump_stack+0x63/0x8b
> [Mon Mar 12 16:17:55 2018]  sysfs_warn_dup+0x5b/0x70
> [Mon Mar 12 16:17:55 2018]  sysfs_do_create_link_sd.isra.2+0x9f/0xb0
> [Mon Mar 12 16:17:55 2018]  sysfs_create_link+0x25/0x40
> [Mon Mar 12 16:17:55 2018]  device_add+0x5b2/0x680
> [Mon Mar 12 16:17:55 2018]  cdev_device_add+0x4e/0x80
> [Mon Mar 12 16:17:55 2018]  ? cdev_init+0x55/0x60
> [Mon Mar 12 16:17:55 2018]  media_devnode_register+0xe7/0x180 [media]
> [Mon Mar 12 16:17:55 2018]  __media_device_register+0x77/0x110 [media]
> [Mon Mar 12 16:17:55 2018]  uvc_probe+0x9dd/0xe60 [uvcvideo]
> [Mon Mar 12 16:17:55 2018]  ? kernfs_activate+0x78/0x80
> [Mon Mar 12 16:17:55 2018]  ? kernfs_add_one+0xe4/0x130
> [Mon Mar 12 16:17:55 2018]  usb_probe_interface+0x14b/0x300
> [Mon Mar 12 16:17:55 2018]  ? uvc_register_video_device+0x150/0x150 [uvcvideo]
> [Mon Mar 12 16:17:55 2018]  ? usb_probe_interface+0x14b/0x300
> [Mon Mar 12 16:17:55 2018]  driver_probe_device+0x310/0x480
> [Mon Mar 12 16:17:55 2018]  __device_attach_driver+0x86/0x100
> [Mon Mar 12 16:17:55 2018]  ? __driver_attach+0xf0/0xf0
> [Mon Mar 12 16:17:55 2018]  bus_for_each_drv+0x6b/0xb0
> [Mon Mar 12 16:17:55 2018]  __device_attach+0xdd/0x160
> [Mon Mar 12 16:17:55 2018]  device_initial_probe+0x13/0x20
> [Mon Mar 12 16:17:55 2018]  bus_probe_device+0x95/0xa0
> [Mon Mar 12 16:17:55 2018]  device_add+0x44b/0x680
> [Mon Mar 12 16:17:55 2018]  ? create_intf_ep_devs+0x5b/0x80
> [Mon Mar 12 16:17:55 2018]  usb_set_configuration+0x514/0x860
> [Mon Mar 12 16:17:55 2018]  generic_probe+0x2e/0x80
> [Mon Mar 12 16:17:55 2018]  usb_probe_device+0x34/0x70
> [Mon Mar 12 16:17:55 2018]  driver_probe_device+0x310/0x480
> [Mon Mar 12 16:17:55 2018]  __device_attach_driver+0x86/0x100
> [Mon Mar 12 16:17:55 2018]  ? __driver_attach+0xf0/0xf0
> [Mon Mar 12 16:17:55 2018]  bus_for_each_drv+0x6b/0xb0
> [Mon Mar 12 16:17:55 2018]  __device_attach+0xdd/0x160
> [Mon Mar 12 16:17:55 2018]  device_initial_probe+0x13/0x20
> [Mon Mar 12 16:17:55 2018]  bus_probe_device+0x95/0xa0
> [Mon Mar 12 16:17:55 2018]  device_add+0x44b/0x680
> [Mon Mar 12 16:17:55 2018]  ? add_device_randomness+0x80/0x110
> [Mon Mar 12 16:17:55 2018]  usb_new_device+0x269/0x490
> [Mon Mar 12 16:17:55 2018]  hub_port_connect+0x70f/0xa90
> [Mon Mar 12 16:17:55 2018]  hub_event+0x850/0xb00
> [Mon Mar 12 16:17:55 2018]  process_one_work+0x1de/0x3e0
> [Mon Mar 12 16:17:55 2018]  worker_thread+0x32/0x410
> [Mon Mar 12 16:17:55 2018]  kthread+0x121/0x140
> [Mon Mar 12 16:17:55 2018]  ? process_one_work+0x3e0/0x3e0
> [Mon Mar 12 16:17:55 2018]  ? kthread_create_worker_on_cpu+0x70/0x70
> [Mon Mar 12 16:17:55 2018]  ? do_syscall_64+0x73/0x130
> [Mon Mar 12 16:17:55 2018]  ? SyS_exit_group+0x14/0x20
> [Mon Mar 12 16:17:55 2018]  ret_from_fork+0x35/0x40
> [Mon Mar 12 16:17:55 2018] **media: media_devnode_register: cdev_device_add failed**
> 03/21/2016
> [Mon Mar 12 16:17:55 2018] Workqueue: usb_hub_wq hub_event
> [Mon Mar 12 16:17:55 2018] Call Trace:
> [Mon Mar 12 16:17:55 2018]  dump_stack+0x63/0x8b
> [Mon Mar 12 16:17:55 2018]  sysfs_warn_dup+0x5b/0x70
> [Mon Mar 12 16:17:55 2018]  sysfs_do_create_link_sd.isra.2+0x9f/0xb0
> [Mon Mar 12 16:17:55 2018]  sysfs_create_link+0x25/0x40
> [Mon Mar 12 16:17:55 2018]  device_add+0x5b2/0x680
> [Mon Mar 12 16:17:55 2018]  cdev_device_add+0x4e/0x80
> [Mon Mar 12 16:17:55 2018]  ? cdev_init+0x55/0x60
> [Mon Mar 12 16:17:55 2018]  media_devnode_register+0xe7/0x180 [media]
> [Mon Mar 12 16:17:55 2018]  __media_device_register+0x77/0x110 [media]
> [Mon Mar 12 16:17:55 2018]  uvc_probe+0x9dd/0xe60 [uvcvideo]
> [Mon Mar 12 16:17:55 2018]  ? kernfs_activate+0x78/0x80
> [Mon Mar 12 16:17:55 2018]  ? kernfs_add_one+0xe4/0x130
[... 13 lines cut ...]
> [Mon Mar 12 16:17:55 2018]  generic_probe+0x2e/0x80
> [Mon Mar 12 16:17:55 2018]  usb_probe_device+0x34/0x70
> [Mon Mar 12 16:17:55 2018]  driver_probe_device+0x310/0x480
> [Mon Mar 12 16:17:55 2018]  __device_attach_driver+0x86/0x100
> [Mon Mar 12 16:17:55 2018]  ? __driver_attach+0xf0/0xf0
> [Mon Mar 12 16:17:55 2018]  bus_for_each_drv+0x6b/0xb0
> [Mon Mar 12 16:17:55 2018]  __device_attach+0xdd/0x160
> [Mon Mar 12 16:17:55 2018]  device_initial_probe+0x13/0x20
> [Mon Mar 12 16:17:55 2018]  bus_probe_device+0x95/0xa0
> [Mon Mar 12 16:17:55 2018]  device_add+0x44b/0x680
> [Mon Mar 12 16:17:55 2018]  ? add_device_randomness+0x80/0x110
> [Mon Mar 12 16:17:55 2018]  usb_new_device+0x269/0x490
> [Mon Mar 12 16:17:55 2018]  hub_port_connect+0x70f/0xa90
> [Mon Mar 12 16:17:55 2018]  hub_event+0x850/0xb00
> [Mon Mar 12 16:17:55 2018]  process_one_work+0x1de/0x3e0
> [Mon Mar 12 16:17:55 2018]  worker_thread+0x32/0x410
> [Mon Mar 12 16:17:55 2018]  kthread+0x121/0x140
> [Mon Mar 12 16:17:55 2018]  ? process_one_work+0x3e0/0x3e0
> [Mon Mar 12 16:17:55 2018]  ? kthread_create_worker_on_cpu+0x70/0x70
> [Mon Mar 12 16:17:55 2018]  ? do_syscall_64+0x73/0x130
> [Mon Mar 12 16:17:55 2018]  ? SyS_exit_group+0x14/0x20
> [Mon Mar 12 16:17:55 2018]  ret_from_fork+0x35/0x40
> [Mon Mar 12 16:17:55 2018] **media: media_devnode_register: cdev_device_add failed**

I am sure that my R200 is healthy since I can run it on MS Windows.

ev-mp, 2018-03-13:
Hi @HumamHelfawi , the log is not informative as it does not shows R200 enumeration events (neither good nor bad).
Is R200 recognized by the host when you run `lsusb`?

HumamHelfawi, 2018-03-13:
Hi @ev-mp after restarting my PC and error message at dmesg -wT disappear. However, I am still getting:

> RealSense error calling rs_create_context(api_version:11201):
>     UVCIOC_CTRL_QUERY:UVC_SET_CUR error 25, Inappropriate ioctl for device

this is the out put of dmesg -wT:

> [Tue Mar 13 16:24:21 2018] usb 2-2: USB disconnect, device number 4
> [Tue Mar 13 16:24:25 2018] usb 2-2: new SuperSpeed USB device number 5 using xhci_hcd
> [Tue Mar 13 16:24:25 2018] usb 2-2: New USB device found, idVendor=8086, idProduct=0a80
> [Tue Mar 13 16:24:25 2018] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [Tue Mar 13 16:24:25 2018] usb 2-2: Product: Intel RealSense 3D Camera R200
> [Tue Mar 13 16:24:25 2018] usb 2-2: Manufacturer: Intel Corp
> [Tue Mar 13 16:24:25 2018] usb 2-2: SerialNumber: SN_2411009530
> [Tue Mar 13 16:24:25 2018] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [Tue Mar 13 16:24:25 2018] uvcvideo 2-2:1.0: Entity type for entity Extension 2 was not initialized!
> [Tue Mar 13 16:24:25 2018] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
> [Tue Mar 13 16:24:25 2018] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [Tue Mar 13 16:24:25 2018] uvcvideo: Unable to create debugfs 2-5 directory.
> [Tue Mar 13 16:24:25 2018] uvcvideo 2-2:1.2: Entity type for entity Camera 4 was not initialized!
> [Tue Mar 13 16:24:25 2018] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.2/input/input36
> [Tue Mar 13 16:24:25 2018] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [Tue Mar 13 16:24:25 2018] uvcvideo: Unable to create debugfs 2-5 directory.
> [Tue Mar 13 16:24:25 2018] uvcvideo 2-2:1.4: Entity type for entity Processing 7 was not initialized!
> [Tue Mar 13 16:24:25 2018] uvcvideo 2-2:1.4: Entity type for entity Camera 6 was not initialized!
> [Tue Mar 13 16:24:25 2018] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.4/input/input37

This is the output of lsusb:

> Bus 002 Device 004: ID 8086:0a80 Intel Corp. 
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 001 Device 005: ID 8087:0a2a Intel Corp. 
> Bus 001 Device 004: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
> Bus 001 Device 003: ID 04f2:b57e Chicony Electronics Co., Ltd 
> Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
> Bus 001 Device 020: ID 046d:0a66 Logitech, Inc. 
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I really thank you for your help.

ev-mp, 2018-03-19:
Hi @HumamHelfawi ,
The error `inappropriate ioctl` somehow indicates that the uvcvideo module is still not functioning properly.
I suggest to switch back to one of the stable Ubuntu kernels (4.4/4.10/4.13) and rerun the application - this will verify the HW chain : USB port/cable/controller. Once it's done then you can switch again to 4.16 kernel and focus on the uvcvideo enablement.

ev-mp, 2018-03-25:
Hello @HumamHelfawi , please update if you need more assistance here

RealSense-Customer-Engineering, 2018-04-10:
[Realsense Customer Engineering Team Comment] 
@HumamHelfawi  Is the issue resolved?

HumamHelfawi, 2018-04-10:
Hi sorry for late response.. No unfortunately, it was not solved out but I am pretty sure the problem from the kernel since there are other software and drivers show similar error (E.g. Virtual Box). you may close the issue. I will try to revert the kernel (I can not right now). I will open new issue if I face further problem.
Thank you very much for your collaboration.

RealSense-Customer-Engineering, 2018-04-11:
[Realsense Customer Engineering Team Comment] 
@HumamHelfawi That's fine. Will close this ticket. And if you still get issue after revert please create new ticket. Thanks!

closed 2018-04-11 by RealSense-Customer-Engineering

shannon112, 2018-12-19:
Hi @ev-mp ,
I also get the same problem on my environment:
Required Info |  -- | --
Camera Model | D435
Operating System & Version | Ubuntu 16.04 LTS
Kernel Version (Linux Only) | (4.10.0-28-generic)
Platform | laptop

uname -a
```
Linux shannon-ThinkPad-X1-Carbon-4th 4.10.0-28-generic #32~16.04.2-Ubuntu SMP Thu Jul 20 10:19:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```
dmesg
```
[Wed Dec 19 23:39:14 2018] usb 2-3: new SuperSpeed USB device number 3 using xhci_hcd
[Wed Dec 19 23:39:14 2018] usb 2-3: New USB device found, idVendor=8086, idProduct=0b07
[Wed Dec 19 23:39:14 2018] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Dec 19 23:39:14 2018] usb 2-3: Product: Intel(R) RealSense(TM) 435
[Wed Dec 19 23:39:14 2018] usb 2-3: Manufacturer: Intel(R) RealSense(TM) 435
[Wed Dec 19 23:39:14 2018] usb 2-3: SerialNumber: 825513025984
[Wed Dec 19 23:39:14 2018] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
[Wed Dec 19 23:39:14 2018] uvcvideo: Unable to create debugfs 2-3 directory.
[Wed Dec 19 23:39:14 2018] uvcvideo 2-3:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
[Wed Dec 19 23:39:14 2018] uvcvideo 2-3:1.0: Entity type for entity Processing 2 was not initialized!
[Wed Dec 19 23:39:14 2018] uvcvideo 2-3:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
[Wed Dec 19 23:39:14 2018] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[Wed Dec 19 23:39:14 2018] input: Intel(R) RealSense(TM) 435 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/input/input16
[Wed Dec 19 23:39:14 2018] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
[Wed Dec 19 23:39:14 2018] uvcvideo: Unable to create debugfs 2-3 directory.
[Wed Dec 19 23:39:14 2018] uvcvideo 2-3:1.3: Entity type for entity Processing 7 was not initialized!
[Wed Dec 19 23:39:14 2018] uvcvideo 2-3:1.3: Entity type for entity Extension 8 was not initialized!
[Wed Dec 19 23:39:14 2018] uvcvideo 2-3:1.3: Entity type for entity Camera 6 was not initialized!
```
lsusb
```
Bus 002 Device 003: ID 8086:0b07 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 138a:0090 Validity Sensors, Inc. 
Bus 001 Device 003: ID 13d3:5248 IMC Networks 
Bus 001 Device 002: ID 8087:0a2b Intel Corp. 
Bus 001 Device 005: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Are there any suggestion?
```