realsenseai/librealsense #1301: Windows 7 viewer.exe crash (and driver install issues)

kwcooper, 2018-03-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

viewpointsa, 2018-03-08:
Required Info | Version
------------ | -------------
Camera Model | D435
Firmware Version | 5.9.2.0
Operating System & Version | Win 10
Platform | PC
SDK Version | v2.10.1

Intel.RealSense.Viewer.exe application crash

``Faulting application name: Intel.RealSense.Viewer.exe, version: 2.7.9.0, time stamp: 0x59be0e69
Faulting module name: ntdll.dll, version: 10.0.14393.1715, time stamp: 0x59b0d8f3
Exception code: 0xc0000005
Fault offset: 0x000709eb
Faulting process id: 0x24c4
Faulting application start time: 0x01d3b6b393c34df1
Faulting application path: C:\Users\mdauphin\Downloads\Intel.RealSense.Viewer.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 771ae2a5-a6af-4f49-9933-ccf152ef498c
Faulting package full name: 
Faulting package-relative application ID: ``

ev-mp, 2018-03-08:
Hello @kwcooper , it is not clear whether you were trying to run Windows 7 (headline) or Windows 10 (template) .
Librealsense SDK is supported on Windows 8.1/10 with MS MediaFoundation driver.
While there is no official support for Windows7 , you may review a related [#1243](https://github.com/IntelRealSense/librealsense/issues/1243) for alternatives

RealSense-Customer-Engineering, 2018-03-20:
[Realsense Customer Engineering Team Comment] 
@kwcooper Did you get the issue on Windows 7 or Windows 10? Please note that there is no official support for Windows 7.

Seen from the log, the SDK version you used seems to be v2.7.9 and not the specified v2.10.1 in the template. If you're trying on Windows 10, please try the latest v2.10.1 in https://github.com/IntelRealSense/librealsense/tree/v2.10.1

RealSense-Customer-Engineering, 2018-03-26:
[Realsense Customer Engineering Team Comment] 
@kwcooper Please update if you need more support. Thanks!

closed 2018-04-03 by dorodnic