realsenseai/librealsense #1302: How to install librealsense on Windows 10?

Alex9812, 2018-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |    Windows 10   | 
| Platform                                 | PC |

### Issue Description

How do I use librealsense with Windows 10? I read the install document (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md), however this is not very helpful.

My goal is to run multiple realsense cameras to obtain a detailed 3D point cloud, and it seems that librealsense is able to do that, however I'm not sure how to make it work.

So far, I have tried to run the multicam example c++ code with visual studio and it just returned with several errors, the first of which says that the header is not read.

ev-mp, 2018-03-08:
Hello @Alex9812 ,
It seem that you actually have several questions that are not necessary related, so I would suggest to start with the HW/SW setup - Camera type used, the SDK version, Windows build so we get a clear picture of the environment.

Additionally, please provide the following:
- Do you build librealsense from the source code or do you use the MS installer.?  
- Did you try to use other examples, such as `realsense-viewer` ?
- Please attach application/console/kernel logs (i.e.  `...the header is not read` ) for the errors you receive.

Alex9812, 2018-03-08:
Hi,

Here is an outline of everything that I have done so far.
1. Connect my realsense camera. Device manager says it is Intel RealSense 410 Depth.
2. Use CMake to build binaries on my desktop of the source code is the folder librealsense-master
3. Open project through CMake
4. Build realsense-viewer under tools.

At this point, I get the following errors (I manually stopped the build after a certain amount of time)
[image]

I'm not quite sure how to move forward from this point? Any advice would be great, I'm pretty much brand new to all of this

zivsha, 2018-03-08:
Hi @Alex9812 ,
We provide a windows installer under the [releases tab: 
](https://github.com/IntelRealSense/librealsense/releases). This is the easiest way to get started, it will install the library along with example projects, and a template for creating a project that uses liberalness

Alex9812, 2018-03-08:
Hi Zivsha,

Thank you for your help. I used the windows installer Intel.Realsense.SDK.exe and ran the hello world program. That seemed to work really well! However I tried to run the "capture" and "pointcloud" examples and I got these errors
[image]

Any advice on solving this issue?

Thanks,

danielw256, 2018-03-08:
To use multiple cameras for depth, doesn't something has to done to prevent IR interference between cameras? I remember someone recommending adding a shaker to one camera, although this seems to only allow 2 cameras. Is this still the best and only way? Or can we expect a hardware synced solution in the future?

Alex9812, 2018-03-08:
So to run the "capture" example, you have to go under tools in Program files and run the application. It just required a little bit of digging around. Thank you for the help everyone.

If possible, I would like to keep this issue open for now. I think it is likely that I may have additional problems when linking two cameras and this issue may be something a lot of beginners like myself could be wondering.

agrunnet, 2018-03-08:
@danielw256 the RealSense D4xx series experience no optical interference, which makes them really good for multicamera usages.  Even though the multiple cameras overlap their field-of-view, there is no information contained in the projected patterns- they only serve to provide texture for stereo matching. So the more texture, the merrier. 
What you are referring to is how people have tried to hack structured light solutions to work, which do interfere with each other since it is imperative that each system knows which projection dots belong to their camera. This has been done by 1. trying to time multiplex, and 2. By shaking one camera. The idea is to make the pattern of the OTHER camera blur out so they just provide a uniform background. This is really impractical as you can imagine. 
Furthermore, Intel is working on releasing official support multicamera HW sync, so each frame is capture at exactly the same time. For many applications this is actually not a requirement and it is fine to just use the time-stamps to align frames. However, for those who want sub-ms sync, that should become available in the not-so-distant future.

danielw256, 2018-03-08:
@agrunnet Thanks for such a good explanation

closed 2018-03-14 by Alex9812