realsenseai/librealsense #1306: Install librealsense on Ubuntu 16.04 with kenerl 4.13.0-36-generic

oPiZiL, 2018-03-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.15.0.0 | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.13.0-36-generic                                 | 
| Platform                                 | PC(NUC)  |
| SDK Version                            |  2.0.2                          | 

### Issue Description
run `realsense-viewer`

**ERROR**
 ```
08/03 16:36:42,600 ERROR [139997495249472] (types.h:178) The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
```
and depth sensor can't be turned on

ev-mp, 2018-03-08:
Hello @oPiZiL ,
Can you be more specific about the environment and installation guide you followed ?
Additionally, please collect and provide the following logs:
- `dmesg -wT` when connecting SR300 to the host
- `v4l2-ctl --list-formats-ext -d /dev/video0`
- `v4l2-ctl --list-formats-ext -d /dev/video1`
(you may need yo adjust the indexes if you have additional UVC devices connected"
- Librealsense version  - 2.0.2 is not a valid SDK version

oPiZiL, 2018-03-09:
1. [dmesg-wT.txt](https://github.com/IntelRealSense/librealsense/files/1795307/dmesg-wT.txt)

2. 
`user@user-desktop:~/librealsense$ v4l2-ctl --list-formats-ext -d /dev/video0`
```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
```
3.
`user@user-desktop:~/librealsense$ v4l2-ctl --list-formats-ext -d /dev/video1`
```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: ''
	Name        : 5a564e49-2d90-4a58-920b-773f1f2
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 2
	Type        : Video Capture
	Pixel Format: ''
	Name        : 495a4e49-1a66-a242-9065-d01814a
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 3
	Type        : Video Capture
	Pixel Format: ''
	Name        : 435a4e49-b602-480f-978c-e4e88ae
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 4
	Type        : Video Capture
	Pixel Format: ''
	Name        : 49564e49-57db-5e49-8e3f-f479532
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 5
	Type        : Video Capture
	Pixel Format: ''
	Name        : 52494150-8536-4841-b6bf-8fc6ffb
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
```
3.`version:        1.1.1.realsense-4.4.0`

oPiZiL, 2018-03-09:
Somehow, I sovled this problem. 
Here is the steps.
1.use `sudo apt-get install realsense-uvcvideo` to get `realsense-uvcvideo` source. I get two version :4.4 & 4.10.0
[Install librealsense ](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)

2.backup directory `realsense-uvcvideo-4.10.0` to HOME directory.

3.remove realsense-uvcvideo by
`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

4.Edit `realsense-uvcvideo-4.10.0/dkms.config`
- change 4.4.0->4.10.0 in line 4
- comment patches
```
PACKAGE_NAME="realsense-uvcvideo"
PACKAGE_VERSION="4.10.0"
MAKE[0]="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build/drivers/media/usb/uvc DRIVER_VERSION_SUFFIX='.realsense-4.10.0' "
CLEAN="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build/drivers/media/usb/uvc clean"
#MAKE[0]="make -C ${kernel_source_dir} -f ${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build/drivers/media/usb/uvc/Makefile M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build DRIVER_VERSION_SUFFIX='.realsense-4.10.0' "
#CLEAN="make -C ${kernel_source_dir} -f ${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build/drivers/media/usb/uvc/Makefile M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build clean"
#PATCH[0]="10-realsense-camera-formats-4.4.patch"
#PATCH[1]="11-uvcvideo-metadata-4.4.patch"
#PATCH[2]="12-uvcvideo-powerlinefrequency-control-fix-4.4.patch"
#PATCH[3]="20-uvcvideo-makefile-include-override.patch"
#PATCH[4]="30-tag-realsense-driver-version.patch"
#PATCH[5]="15-realsense-camera-formats-4.10.patch"
#PATCH_MATCH[0]="4.4"
#PATCH_MATCH[5]="4.10.[0-9]"
BUILT_MODULE_NAME[0]="uvcvideo"
BUILT_MODULE_LOCATION[0]="drivers/media/usb/uvc"
DEST_MODULE_LOCATION[0]="/kernel/drivers/media/usb/uvc"
AUTOINSTALL="yes"
#Replacing the activelyloaded module with post-install doesn't work correctly - requires further investigation
#The next two lines are leave for future refinement
#POST_INSTALL="reload_kernel_module.sh"
#POST_REMOVE="reload_kernel_module.sh"
```
5.patch `realsense-uvcvideo` mannually
cd into `realsense-uvcvideo-4.10.0`
`sudo patch -p1 < ./patches/15-realsense-camera-formats-4.10.patch`
`sudo patch -p1 < ./patches/20-uvcvideo-makefile-include-override.patch`
`sudo patch -p1 < ./patches/30-tag-realsense-driver-version.patch`

6.copy `realsense-uvcvideo-4.10.0` to /usr/src

7.`sudo dkms build -m realsense-uvcvideo -v 4.10.0`
   `sudo dkms install -m realsense-uvcvideo -v 4.10.0`
8.`sudo apt-get install librealsense2-utils`
9.`reboot` and run `realsense-viewer`

ev-mp, 2018-03-09:
Hello @oPiZiL and thanks for the detailed update!

You may continue to use the DKMS driver with all the work you did. Alternatively, you can switch tothe [updated DKMS package](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#linux-distribution) that was published as part of the the latest v2.10.1 release.

In addition to improvements, bug fixes and optimizations, the new DKMS also supports the latest LTS kernel 4.13 that is gradually promoted to Ubuntu distributions.

In case you chose the upgrade path - the DKMS package name was also [modified](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features--improvements) to be consistent with the rest of the SDK.

ev-mp, 2018-03-12:
Hello @oPiZiL , please let me know if you need further assistance with this issue.

oPiZiL, 2018-03-14:
@ev-mp  thanks for help. I have solved my problem.

closed 2018-03-14 by oPiZiL