realsenseai/librealsense #1308: Patch Failure (realease 2.10.1, kernel 4.4.0-116)

marvin-hao, 2018-03-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | | 
| Operating System & Version |   Linux Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.4.0-116                                         | 
| Platform                                 | PC|
| SDK Version                            |  2.10.1                           | 

### Issue Description
encountered error while patching (from dmesg):

videodev: version magic '4.4.0-116-generic SMP mod_unload modversions ' should be '4.4.0-116-generic SMP mod_unload modversions retpoline '

The error displayed in the patching session was:

Handle videodev :
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Exec format error
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded

It was working with 4.4.0-112 kernel

marvin-hao, 2018-03-09:
patch worked when I boot with 4.4.0-112 kernel

marvin-hao, 2018-03-09:
This issued might be caused by inconsistency between 116 kernel distribution and src code.

closed 2018-03-09 by marvin-hao

RealSense-Customer-Engineering, 2018-03-14:
[Realsense Customer Engineering Team Comment] 
Hi @marvin-hao,
still need any further help?