realsenseai/librealsense #1309: How to create volumetric with four D435

kdoonboli, 2018-03-09:
How can I create volumetric footage with four D435 cameras?

Similar to how it’s done here:
https://youtu.be/9oQDz_cfUlo

zivsha, 2018-03-09:
I have no real knowledge on how to do this, and you could probably find better answers from others, but basically you need to create four point clouds and make sure they all use a single coordinates system.
Since the SDK provides a way to create a point cloud from the depth data, (I think) that all that's left is to align all point clouds to a single one. To transform a depth point from one coordinates system another you can perform the same process we do when aligning depth and color, by using the camera's intrinsic data and the extrinsics between the sensors. I don't know how you can calibrate two depth sensors on different cameras but you can probably guestimate the rotation and translation between the two until you get it right.
Again, I might be too naive thinking this is as "simple" as that but it's probably a good start.

kdoonboli, 2018-03-30:
@RealSense-Customer-Engineering That was very helpful. Three questions: 1) where is the link to the white paper? It's not in the page you provided.  2) does this totally remove the need for a green screen? or for better results, it's better to have a green screen? 3) it says each computer was connected to one dedicated PC with i7 processor. Why? Shouldn't a quad-core CPU with 4 dedicated USB controllers (5gb each) be able to handle the load?

Thank you!

RealSense-Customer-Engineering, 2018-03-31:
[Realsense Customer Engineering Team Comment] 
I made a mistake to state the white paper was attached. The white paper is available from your Intel account manager and you will need a signed NDA. With respect to the green screen, that is a subjective judgement; the green screen does perform better but it all depends on your requirements. The mention of the PC is only describing the equipment used for the demo and shows that it is possible to use a single PC. The mention of i7 is important because the load is not just the USB bandwidth but the processing of the incoming data.

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
Hello @kdoonboli, the white paper has been made public now at https://realsense.intel.com/intel-realsense-downloads/.

TheMikeyR, 2018-04-17:
@RealSense-Customer-Engineering You are referring to "Multiple Camera WhitePaper" right? It says revision 2.0 on website but rev 0.4 in the document linked on the website.

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
@TheMikeyR, I don't know who edited the web site but the 0.4 paper is the most recent (actually the only published rev).