realsenseai/librealsense #1310: difference between depth colormap using python and rs-align tool

adarvit, 2018-03-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300  | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.10.1 }                          | 

### Issue Description
<question>

I noticed that when i am using your example for alignment frames with python, the colormap depth looks different from the colormap frames using realsense viewer or rs-align example tool. what could be the reason for that? is there a way to get the same colormap using python?

zivsha, 2018-03-09:
Hi @adarvit ,
The `rs2::colorizer` class that performs the colorizing of depth in the library performs histogram equalization of the depth frames and sets the color accordingly (and with different min-max range) whole the python version (if I remember right) uses opencv function to normalize the depth values within a given range and map the pixel's color according to the color map you provide, witout perforing histogram equalization. The color maps are the same but the choice of color per depth value is calculated differently

adarvit, 2018-03-10:
Thank you very much for you reply

closed 2018-03-10 by adarvit