realsenseai/librealsense #1319: Playback functions when rs2::pipeline is configured to playback from file

limgm, 2018-03-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.10.1 | 

Hi 

Following the example from Realsense Record and Playback https://github.com/IntelRealSense/librealsense/blob/master/src/media/readme.md

I am able to configure the rs2::pipeline to play a streaming session and the playback would repeat with the recent addition of playback repeat to pipeline with device from file in SDK 2.10.1.

I would like to incorporate other functions of playback (pause, resume, seek ...) similar to those playback options available in realsense-viewer GUI. May I ask if it is possible and any examples on how use the functions of playback when rs2::pipeline is configured to playback from file?

Thank you.

zivsha, 2018-03-10:
The playback options you are referring to work at the device level. So you need the device in order to control the playback. Since the pipeline is not extendable to playback functionality, you need to ask it for the device it is using. To do that, you can either take the return value from `auto profile = pipeline.start(...)` or from `auto profile = pipeline.get_active_profile()` and then:
```cpo 
auto dev = profile.get_device();
if(auto p = dev.as<rs2::playback>())
{
    //Use p for playback functionality
}
```
I'm not sure how the pipeline will take the `pause` function since there is a timeout for `wait_for_frames` so make sure you handle that.

limgm, 2018-03-11:
Hi @zivsha 

Thank you for your help! 
Using your example, I am able to control playback using the return value from rs2::pipeline_profile.

To prevent the timeout for wait_for_frames, use poll_for_frames instead and the pipeline is able to take the pause function =)

closed 2018-03-11 by limgm