realsenseai/librealsense #1320: Realsense D435 and Aero Compute Board

mgildner, 2018-03-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.8.15.0 | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.4.76-aero-1.2 | 
| Platform                                 | Intel Aero Compute  |
| SDK Version                            |  2.0| 

### Issue Description

I followed these instructions [https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation](https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation) to install ubuntu on the compute board. I installed 16.04 with kernel version 4.4.

I then proceeded to try to install the Realsense 2.0 SDK.  I tried installing via these instructions [https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md).  The installation completes but with errors related to the uvcvideo kernal patch. When I run the realsense-viewer I get dropped frame errors and no depth images.  The stereo stream is the only one available.  I am plugging in the realsense through a powered USB 3.0 Hub capable of 900ma per port.  

I followed up trying to build from source.  After uninstalling the previous packages, I proceeded to follow these instructions [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).  That completes but does not show an updates to uvcvideo when running `sudo dmesg | tail -n`.  Same errors when I try to run realsense-viewer.

I am thinking the issue might be kernel patching issue since the `aero-system` package modifies the kernel as well. `uname -r` reports the modified 4.4 intel aero kernel.

At this time, is it possible to use the D400 series realsense modules with the aero compute board?

RealSense-Customer-Engineering, 2018-03-12:
[Realsense Customer Engineering Team Comment] 
Hi @mglidner,

Do you connect D435 in the USB3.0 OTG port?
Can you run "lsusb -t" and "rs-enumerate-devices" to show the log?
When you run 'realsense-viewer', can you also run "dmesg -wT" to show the log for the error you mentioned?

mgildner, 2018-03-21:
I did connect D435 to the USB3 OTG port through a USB3 OTG Cable (this one | Firmware Version | 05.09.02.00 |) and a USB 3.0 hub (tried both of these https://www.amazon.com/gp/product/B005AS337Y/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1, https://www.amazon.com/gp/product/B0725Z7PJT/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1).  

D435 works on both hubs work successfully on a windows machine.

This is what I get on "lsusb -t"
`/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/7p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
`
As you can see the realsense will not even show up despite being plugged in.  On some hubs it shows up, but always on Bus 01 as a high speed device.

RealSense-Customer-Engineering, 2018-03-21:
[Realsense Customer Engineering Team Comment] 
Hi @mgildner
Still like the USB signal quality issue which causes D435 can't be showed or recognized a USB3 device. You can try to provide the external power to your USB3 hub and use a USB3 qualified/short cable.

mgildner, 2018-03-21:
Hi, can you confirm that the USB C issue mentioned here might be the same problem.

The intel compute board also has a ASCI that switches the MUX from what I understand.  

Is this going to require a hardware revision of the D435?

https://github.com/IntelRealSense/librealsense/issues/1391

daversintel, 2018-03-21:
Try to replace the uvcvideo patch script with the one (you need to replace the file extension and make it executbale in order to run it.) attached to see if it helps.
[patch-realsense-ubuntu-xenial.txt](https://github.com/IntelRealSense/librealsense/files/1834795/patch-realsense-ubuntu-xenial.txt)

I boot the Ubuntu in Aero kernel, and here are the step-by-step instructions to make D435 work on Aero Compute Board. You might need to disconnect and reconnect DS435 several times if it's not recognized.

1.	Install aero- system on the Ubuntu customer already installed:
echo 'deb https://download.01.org/aero/deb xenial main' | sudo tee /etc/apt/sources.list.d/intel-aero.list
wget -qO - https://download.01.org/aero/deb/intel-aero-deb.key | sudo apt-key add -
sudo apt-get update
sudo apt-get upgrade
sudo apt-get -y install gstreamer-1.0 libgstreamer-plugins-base1.0-dev libgstrtspserver-1.0-dev gstreamer1.0-vaapi gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-libav ffmpeg v4l-utils
sudo apt-get -y install aero-system
sudo reboot 
2.	Get aero version and update BIOS
       sudo aero-get-version.py
       sudo aero-bios-update
       sudo reboot (Use advanced option is necessary to boot to aero kernel. ‘uname -r’ should show “4.4.76-Aero-1.2” after reboot.)
3.	Update FPGA
       sudo jam -aprogram /etc/fpga/aero-compute-board.jam
4.	Update system
       sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
       sudo apt-get install --install-recommends linux-generic-lts-xenial xserver-xorg-core-lts-xenial xserver-xorg-lts-xenial xserver-xorg-video-all-lts-xenial xserver-xorg-input-all-lts-xenial libwayland-egl1-mesa-lts-xenial
       sudo update-grub && sudo reboot
5.	Install packages needed for librealsense
        sudo apt-get install git cmake
               sudo apt-get install libusb-1.0-0-dev pkg-config libgtk-3-dev
                sudo apt-get install libglfw3-dev
6.	Get librealsense source code and build it:
         git clone https://github.com/IntelRealSense/librealsense.git (Pick a folder for the code).
          Navigate to librealsense root directory and run mkdir build && cd build
          cmake ../ -DBUILD_EXAMPLES=true
    sudo make uninstall && make clean && make && sudo make install    
7.	Replace the “patch-realsense-ubuntu-xenial.sh” with the modified version (attached) in the librealsense root directory:
         ./scripts/patch-realsense-ubuntu-xenial.sh
         Connect the camera and Reboot
8.	Try if DS5 is recognized by Aero Compute Board:
          ls /dev/video* should show /dev/video0, /dev/video1, …,  /dev/video12
          rs_enumerate-devices should show DS5 device
          rsrealsense-viewer should see depth map
9.	Here are the web sites for reference:
a.	https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation
b.	https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
c.	https://github.com/IntelRealSense/librealsense

mgildner, 2018-03-22:
Thank you for this information.  After working through the steps, the patch update script is having some issues.  First, the script uses a blank string for the kernel branch when booting with the aero kernel.  To get around this I just manual updated this string as 'master'.  This allows the patch script to find the right patch files.  However once it gets to try and patch the uvnvideo module it hangs.

mgildner, 2018-03-22:
Also I received an update on the Intel Aero side.  See this thread [here](https://communities.intel.com/message/533400#533400).  It sounds like the it is not likely that we can get SuperSpeed USB3 working on the compute board without a yet to be made available cable from intel.  

Were you able to find a compatible cable?  Or were you just using the D435 at highspeed USB speeds?

daversintel, 2018-03-22:
1. You might need to find a right USB-OTG3 cable.
2. It sounds like there is still a problem with the UVC Video patch script. The script can be run line-by-line, and it can help you to isolate the problem.

ioreed, 2018-03-23:
We have a similar problem on the UP Board. I ordered an [OTG patch cable](https://up-shop.org/up-peripherals/22-usb30-otg-cable.html) from their site that might fix this. Will post results on the similar UP Board thread [#1391](https://github.com/IntelRealSense/librealsense/issues/1391).

RealSense-Customer-Engineering, 2018-04-12:
[Realsense Customer Engineering Team Comment] 
Hi @mgildner,

Is everything going well to have a right cable?

ChrisHughes, 2018-04-15:
Just a heads up, we are deploying 3 Aero compute boards and have resolved the cable issue by removing the aero board's heat sink to expose access to the USB 3.0 micro type AB header, and putting a blob of solder on the 'sense pin' bridging it with ground, so connecting pins 4 and 5. We then can connect the camera via a USB 3.0 hub with a 3.0 micro B connector using this cable https://www.digikey.com/product-detail/en/qualtek/3023007-01M/Q551-ND/2356457.
Very few cables come with this pin connected, even the qualtek one linked above did not have that internal connection on the A type plug.

See https://gct.co/images/products/usb-connector/USB-3-A-AB.jpg for details of this standard.

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
Hi@mgildner,

If no other issue, will close this.

closed 2018-05-31 by RealSense-Customer-Engineering

JenaEmz, 2018-12-25:
@ChrisHughes Could you tell me which pin in your picture is the "sense pin"

ChrisHughes, 2018-12-25:
@JenaEmz see this picture for claification http://pinoutguide.com/visual/gen/micro_usb_3_connector.jpg note that the diagram shows a male connector. Pins 4 and 5 need to be connected on the aero board - this grounds the OTG ID line.

JenaEmz, 2018-12-29:
@ChrisHughes Thank you very much :D