realsenseai/librealsense #1323: Filtering IR images before depth calculation

vladsterz, 2018-03-12:
----------------------------------------------------------------------------------------------------

Required Info |  
-- | --
Camera Model | D415 and D435
Operating System & Version | Win 10
Platform | PC
SDK Version | 2.10.1

### Issue Description
Hello once again. I have a question about the library/devices. Is it possible to mess around with the IR images and then create disparity? Let's say I want to filter the IR images and I want to create depth from the filtered IR frame couple. Is there a way to do it?

Thanks

agrunnet, 2018-03-12:
No, sorry, not the way I believe you want to. 
Clearly the chip can control the "IR images" via gain and exposure and frame rate, but you cannot in software apply some filters to these images BEFORE they are sent to the stereo calculation portion of the chip.

However, you can tell the ASIC to give you disparity instead of depth, but that is not what you want.

vladsterz, 2018-03-13:
Yeah. I've noticed that IR images contain some "white-ish" spatial noise and I thought if I could get rid of it, I could get depth measurements with less fluctuations. Sad to hear that it's not possible.

closed 2018-03-28 by dorodnic