realsenseai/librealsense #1324: win10 D435 always crash after a few seconds

timyy, 2018-03-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.9.2.0  |  
| Operating System & Version |  Win 10/ Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  PC   |
| SDK Version                            |   legacy / 2..                          | 

### Issue Description
I buy two intel realsense D435 , upgrade to 5.9.2.  I use viewer to view deep and rgb,  have recorded to bag about 10 seconds, it seems good.
 but after that , the viewer crashed every time after run a few seconds , i must restart computer, but the viewer crash again.  my computer is dell xps8930, windows 10 is update. nvidia driver 391. it is a new system.   the viewer show the device down.

ps: the driver in the intel support site "Intel RealSense D400 Series UWP Driver v5.1.160.5.zip" has a auto dfu service, it upgrade my d435 bios, this design is very offensive

ev-mp, 2018-03-13:
Hello @timyy , can you provide some details :
- What is the SDK version you're using. Is it precompiled binaries or built from source code.
- Do you connect both D435 to the same usb port ?
- Error messages received from the `realsense-viewer`
- What do you mean by  "..the viewer show the device is down"

zzhsmt, 2018-03-19:
I have the same question, intel realsense D415. I use viewer to view deep and rgb, have recorded to bag about 2 seconds. 
previous：
[image]

after：
[image]

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
Hello @timyy,

would like to confirm if you still need support on this topic?
thanks.

hi @zzhsmt,
do you run anything between your previous and after? update windows, driver or camera FW?
thanks.

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
Hello @timyy,
(1) there are similar issues that cause by the cable quality.  you can see if shorter cable or another cable can help fix this issue

(2) is both D435 connected to same USB3 port through Hub and streaming simultaneously with high resolution and fps, with higher resolution and fps, D435 required higher  bandwidth

(3) is there any mouse plugged? if not, please see if plug a mouse can help