realsenseai/librealsense #1335: 4.14 kernel support

manusovich, 2018-03-13:
When i am trying to execute script below during lib installation, getting back error that 4.14 kernel is not supported
```
./scripts/patch-realsense-ubuntu-xenial.sh
```

ev-mp, 2018-03-13:
Hello @manusovich , the officially supported kernels for Linux are Ubuntu LTS versions 4.4, 4.10 and 4.13.
Please see related request [#1294](https://github.com/IntelRealSense/librealsense/issues/1294) .

Please note that in order to stream Depth and Infrared data for D400 and SR300 you actually can proceed without applying any patches. The patches are required for certain formats (e.g. synthetic color for D415) and advanced features, such as Metadata attributes.

RealSense-Customer-Engineering, 2018-03-15:
[Realsense Customer Engineering Team Comment] 
Hi @manusovich,
Still need any help since the general formats for D400 and SR300 are supported in 4.14?
If you still need to support certain formats, you can still modify the patches/code to patch for 4.14.

RealSense-Customer-Engineering, 2018-05-23:
[Realsense Customer Engineering Team Comment]
I will close this since the D400 depth format is supported in 4.14. If any concern, you still can open it.

closed 2018-05-23 by RealSense-Customer-Engineering