realsenseai/librealsense #1336: LabVIEW Wrapper issues error1097

RedApach, 2018-03-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out Wrapper LabVIEW
Helloword point cloud/ Helloword depht and color
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
yes
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | () | 
| Operating System & Version |   {Win /10)64bit  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/  |
| SDK Version                            |  { legacy / 2.0v labview wrapper 2-8-}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I have a D435 camera on PC win 10 64bit USB3. I am able to use the viewer with all the fonctionnality of the cammera. I want to use the LabVIEW wrapper. Exemple Hello Word Lefright Ip works well but I am not able to execute Depth and color VI. When I execute this VI I  get the error 1097 on startall VI. It is working if i change the count of streams to one ( but i get only the depth map)
REgards

RedApach, 2018-03-15:
Hello, 
Thanks for you reply.
Did you try this Sample. Without changing anything I get the error 1097.

agrunnet, 2018-03-15:
Correct. The demo you are using is for D415. 
We will look at posting a new demo for the D435 as well, very shortly.

You have to select input stream an depth + monochrome (not Color). And when you read the frames you have to handle them as monochrome.

agrunnet, 2018-03-15:
1. First change the ConfigAll.vi to have input "Depth+L Monochrome". This will tell the depth camera to send depth and monochrome images. 
[image]
2. Then right mouse click on case structure for interpreting the frames and "duplicate case". Change the color selector input to "T".  This will allow you to interpret the frames as monochrome images when they are captured. 
[image]
3. Finally, when displaying the monochrome images, you want to display as mono, not color. So change the T to F.
[image]

agrunnet, 2018-03-15:
Here is the VI where I did all this hard work for you :)

[RealSense_HelloWorld_Depth_and_Monochrome.zip](https://github.com/IntelRealSense/librealsense/files/1816040/RealSense_HelloWorld_Depth_and_Monochrome.zip)

agrunnet, 2018-03-15:
[image]

agrunnet, 2018-03-15:
Here is a LabView 2016 zipped folder. Also uses latest LibRealSense dll. (version 2.10)
[labview_samples_210.zip](https://github.com/IntelRealSense/librealsense/files/1816120/labview_samples_210.zip)

RedApach, 2018-03-15:
Thanks a lot. I am trying right now ;)

RedApach, 2018-03-15:
It is possible to provide the  LibRealSense dll. (version 2.10)? Where do you git it? it is not in the sdk. 
Regards

agrunnet, 2018-03-15:
The Dll was gotten by compiling the current libRS to a target of Release and x64. I then just copied the dll to the Labview vi library. In general this should work and should not break anything in Labview for all new versions of the dll as well.

RedApach, 2018-03-16:
Hello, 
I do not found the DLL on the .zip that you attached. It is possible to send me the DLL?
Thanks a lot

agrunnet, 2018-03-16:
[realsense2.zip](https://github.com/IntelRealSense/librealsense/files/1819461/realsense2.zip)

RedApach, 2018-03-16:
Hello argunet and thanks a lot for your time. 
Your VI doesn't solve the issue. I still get the error Error 1097.

RedApach, 2018-03-16:
No sorry, it is working weel. Thanks a lot. an other issues at the sames time ( USB 2 port... :( ) 
Thanks a lot for your time and your reactivity

rullyn, 2018-03-28:
Hello, 
I also face the same problem (error 1097) can i use that code for SR 300?
Thank you.

RedApach, 2018-04-17:
Hello, 
Did you work on the D435 Version, It is possible to do a sample for the POINTCLOUD example?

agrunnet, 2018-04-20:
Yes. Will make one for you...

agrunnet, 2018-04-21:
Here is a whole new package. Please let me know if it works for you. 
I added examples specifically for D435 which uses monochrome stereo imagers.
It is also based on the last LibRS 2.4.10. The pointcloud examples work with the D435.
Password is intel123
[RealSense SDK2.10.4 for LabVIEW.zip](https://github.com/IntelRealSense/librealsense/files/1935035/RealSense.SDK2.10.4.for.LabVIEW.zip)

agrunnet, 2018-04-22:
Here is a Labview15 version for those who dont have Labview 2017. 
PointCloud demo only.  There is also a separate zipped realsense2.dll for 32bit systems in case you have that. Just replace the DLL. 
[PointCloudDemo_for_LabVIEW15.zip](https://github.com/IntelRealSense/librealsense/files/1935837/PointCloudDemo_for_LabVIEW15.zip)
 
Password: intel123
[realsense2_32bit.zip](https://github.com/IntelRealSense/librealsense/files/1935836/realsense2_32bit.zip)
 
and here is the 64bit DLL in case you need it. 

[libRealSenseDLL_64bit.zip](https://github.com/IntelRealSense/librealsense/files/1935852/libRealSenseDLL_64bit.zip)
same password.

RedApach, 2018-04-24:
Hello, Thanks a lot for your support. 
Last question  : it is possible to use your wrapper on a linux RT target?

agrunnet, 2018-04-24:
I have not tried it. In priciple yes but you will need to compile the .so library and wrap it in Linux I believe.

closed 2018-05-14 by RealSense-Customer-Engineering