realsenseai/librealsense #1362: Intel Real Sense D400 series Color Camera Module - Hardware

Fragennari, 2018-03-14:
Missing documentation in `Intel Real Sense Camera 400 series (DS5) Product Family Datasheet`about the color sensor custom added to D410 and D430 cameras. 
Looking for documentation and suppliers for color sensor and adapter/interposer between sensor and board.
[image]

RealSense-Customer-Engineering, 2018-03-20:
[Realsense Customer Engineering Team Comment] 
hi @Fragennari,
please refer to below link for latest datasheet, and we are not selling standalone interposers/color sensors, so would suggest you to use D415/D435 for RGB with depth.
https://www.intel.com/content/www/us/en/support/articles/000026827/emerging-technologies/intel-realsense-technology.html

RealSense-Customer-Engineering, 2018-05-02:
[Realsense Customer Engineering Team Comment]
hi @Fragennari,

do you still need support on this topic?
thanks.

Fragennari, 2018-05-02:
No, Thanks!

closed 2018-05-02 by Fragennari