realsenseai/librealsense #1364: How to get realsense's log ?

zenving, 2018-03-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description

How to enable realsense's log, and get the log's data ? THX.

RealSense-Customer-Engineering, 2018-03-14:
[Realsense Customer Engineering Team Comment] 
Hi @zenving,

You can review below Q&A link.
For example (Windows): set LRS_LOG_LEVEL=DEBUG, use the same command window to run the app. Then, you can get the log file (*.log) in that directory.
https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A

dorodnic, 2018-03-22:
Closing this issue, @zenving feel free to open new ones if you have additional questions regarding camera troubleshooting

closed 2018-03-22 by dorodnic