realsenseai/librealsense #1365: Compare and compute distance between two point clouds

Alex9812, 2018-03-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> v2.10.1 | 
| Operating System & Version |   {Win 10}  | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.10.1}                          | 

Does librealsense provide the functionality to compare a reference frame of a point cloud with any subsequent frames? The camera is held constant the whole time.

I want to see deformation of an object, is there a way I can do this?

agrunnet, 2018-03-15:
No such function exists in the API. But it wouldn't be hard to do I would think, provided the camera is fixed. Just subtract one depth map from another.

Alex9812, 2018-03-15:
I attempted to do that however I am not sure how the point map is stored. Can I assign a specific frame to a certain variable and subtract any other incoming frames with this reference frame?

I am using the rs-pointcloud example as a basis.

agrunnet, 2018-03-15:
I would recommend doing the comparison on the depth map first and then converting to point cloud afterwards. This would allow you to identify identical points  as coming from the same (x,y) location and compare their z.
 
Otherwise you have to come up with a more complex method for figure out what point in 3D space at t=0, belongs to which point at t=t1.

Alex9812, 2018-03-15:
Good point, but how do I access the depth values? From the rs-pointcloud example, the depth frame is captured by the command frames.get_depth_frame(). However I am unsure how to access this information. Is this information stored as an array?

Alex9812, 2018-03-15:
Any help on this?
I've tried to subtract one depth frame from another but the result creates an int value instead of a rs2 value, which cannot be mapped using draw_pointcloud

jjoachim, 2018-03-15:
rs2::frame has a "const void* get_data() const" function (hpp/rs_frame.hpp:337). Maybe that's what you want?

agrunnet, 2018-03-15:
I am not sure whether you want to visualize in that way by simply subtracting. 
If you want to visualize a change on the point cloud and highlight/show depth that changed, you probably just want to do the diff calculation and then case statement if any point is above a certain diff threshold deltaZ you would show those points on the point cloud in their new location. 
If you show all diffs on point cloud then everything collapses to be a few mm away from you.

RealSense-Customer-Engineering, 2018-04-16:
[Realsense Customer Engineering Team Comment] 
@Alex9812
Does the problem still exist? Can it be closed?

closed 2018-05-02 by RealSense-Customer-Engineering