realsenseai/librealsense #1372: test

RobO-Intel, 2018-03-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2018-04-18:
@RealSense-Customer-Engineering 
Do we need to keep this open?

JasonChang13, 2018-05-08:
need some test for now

RealSense-Customer-Engineering, 2018-05-08:
[Realsense Customer Engineering Team Comment]
test message 2 from Zendesk

RealSense-Customer-Engineering, 2018-05-08:
[Realsense Customer Engineering Team Comment]
test message 3

RealSense-Customer-Engineering, 2018-05-08:
[Realsense Customer Engineering Team Comment]
test 4

RealSense-Customer-Engineering, 2018-05-08:
[Realsense Customer Engineering Team Comment]
test 5

RealSense-Customer-Engineering, 2018-05-08:
[Realsense Customer Engineering Team Comment]
test 6

RealSense-Customer-Engineering, 2018-05-08:
[Realsense Customer Engineering Team Comment]
test 7

RealSense-Customer-Engineering, 2018-05-08:
[Realsense Customer Engineering Team Comment]
test 8

RealSense-Customer-Engineering, 2018-05-17:
[Realsense Customer Engineering Team Comment]
segment broamarket
type question
priority low
issue category question
summary test
product D400
OS windows
language=others

JasonChang13, 2018-05-17:
ignore me

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
'test #1372'

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
"test #1372"

JasonChang13, 2018-07-09:
`#1372`

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
<code>
'#1372'
''#1372''
</code>

JasonChang13, 2018-07-12:
`line 1
line 2
line 3`

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
<code>
line 1
line 2
</code>

closed 2018-08-09 by RealSense-Customer-Engineering