realsenseai/librealsense #1374: Writing an averaging filter

jjoachim, 2018-03-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.2                          | 

I'm having issues with writing a simple frame averaging filter. Normally (for other devices) I cast data pointers into opencv (cv::Mat points directly to data) and then do my averaging there. The issue here is that creating a new synthetic frame requires all of these fancy synthetic devices. Somewhere along the way, I'm getting a bad reference and segfaulting (memory elsewhere is being corrupted, and valgrind has been unhelpful in this case). I would post my code directly, but there's a lot going on and need time to produce a minimal example.

Are there any examples of something similar that I can look at to get a better bearing on it? I'm wondering if I shouldn't try to do it directly with processing blocks. somehow.

vladsterz, 2018-03-19:
I guess it is for the depth sensor. You can always work inplace on the data and/or  copy there.
You can create a rs2::frame and copy the processed data to the data pointer.

dorodnic, 2018-03-19:
Hi @jjoachim 

If you want to perform processing within SDK processing pipe, you can wrap your code in a processing block like this: 
```cpp
rs2::frame_queue q;
rs2::processing_block pb([](rs2::frame f, rs2::frame_source& src) {
    rs2::frame new_frame = src.allocate_video_frame(f.get_profile(), f);
    rs2::video_frame orig = f;
    auto ptr = (uint16_t*)new_frame.get_data();
    memcpy(ptr, f.get_data(), orig.get_width() * orig.get_height() * orig.get_bytes_per_pixel());
    // Put your processing code here
    src.frame_ready(new_frame);
});
pb.start(q);
```
And then invoke it, like so: 
```cpp
pb.invoke(depth);
depth = q.wait_for_frame();
```

Let me know if this helps, or if you encounter any problem

jjoachim, 2018-03-19:
That deep copy example gives enough insight for me to do what I need. Thank you!

RealSense-Customer-Engineering, 2018-03-23:
[Realsense Customer Engineering Team Comment] 
hi @jjoachim,

Do you need more support on this topic or we are ok to close this ticket?
thanks.

closed 2018-03-23 by jjoachim