realsenseai/librealsense #1377: rs-software-device example doesn't track bytes_per_pixel properly

jjoachim, 2018-03-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00) | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            | 2.10.2                     | 

While debugging my own issues, I noticed that get_bytes_per_pixel() gives 0 for synthetic frames. In the rs-software-device example, I modified the code a bit to the following to reproduce the issue:

```
if (auto as_depth = depth.as<rs2::depth_frame>()) {
  std::cout << as_depth.get_bytes_per_pixel() << std::endl;
  points = pc.calculate(as_depth);
}
```

I noticed that in the function on_video_frame(), if you multiply the input BPP by 8 you get the correct answer (which is 2). It's as if the input was expecting bits_per_pixel. The rs2_software_video_frame struct, which is the input to on_video_frame, has a data member bpp. The example uses this as if it were bytes_per_pixel, but does it actually mean bits_per_pixel?

zivsha, 2018-03-16:
Hi @jjoachim ,
Looks like a bug in: https://github.com/IntelRealSense/librealsense/blob/57e623cd7d5f3912ffae5af86e104d93577fedb0/src/software-device.cpp#L131
Should be `software_frame.bpp * 8` I think

dorodnic, 2018-03-18:
Thank you, @jjoachim 
We will try to fix ASAP, feel free to open a pull-request if you would like.

dorodnic, 2018-03-24:
Should be resolved in `v2.10.2`, thank you for bringing-it up. Please feel free to open new issues if you still encounter any problem

closed 2018-03-24 by dorodnic