realsenseai/librealsense #1380: Bug in distortion model

UshakovSergeySergey, 2018-03-16:
I was looking at the library's implementation of the distortion model and noticed some strange thing.
Here is the link to the source code:
https://github.com/IntelRealSense/librealsense/blob/5e73f7bb906a3cbec8ae43e888f182cc56c18692/include/librealsense2/rsutil.h#L15

Look at how you project and deproject point into image, here are these methods
rs2_project_point_to_pixel
rs2_deproject_pixel_to_point

I have noticed that when we deproject pixel to 3d point you are applying undistort method (in the example there is Brown conrady model used).
But when we projecting 3d point back to pixel instead of distorting uv's you are undistorting them once again.

Did I missed something or maybe I don't understand something?
Lines 24-31 and 56-61 are equivalent. The common sense tells me that the code in lines 56-61 must UNDISTORT uv coordinates and lines 24-31 must DISTORT these coordinates. But they do the same thing - UNDISTORT.

RealSense-Customer-Engineering, 2018-06-01:
[Realsense Customer Engineering Team Comment]
hi @UshakovSergeySergey,

I checked release note seems no fix for this question yet, but i also checked the rsutil.h, line 24-31 and 56-61 are not equivalent now, would like to know if your issue still there?
thanks.

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
hi @UshakovSergeySergey,

would like to confirm if your issue still exist?

closed 2018-06-27 by RealSense-Customer-Engineering