realsenseai/librealsense #1381: SR300 - depth - duplicated objects in-front of the camera

kmonachopoulos, 2018-03-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Firmware Version                   | (Uknown) | 
| Operating System & Version |   {Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  (4.4.0-116-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { Uknown }                          | 

### Issue Description

I am trying to read only the depth information (ideally as a 2D greyscale image) using opencv. I have managed to plot the camera feed image but I am getting duplicated objects in-front of the camera.

Code : 

```
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API

int main(int argc, char * argv[]) try
{

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
    pipe.start();

    using namespace cv;
    const auto window_name = "Display Image";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    while(cvGetWindowHandle(window_name))
    {
        rs2::frameset frames = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        //rs2::frame ir_frame = frames.first(RS2_STREAM_INFRARED);
        rs2::frame depth_frames = frames.get_depth_frame();

        // Query frame size (width and height)
        const int w = depth_frames.as<rs2::video_frame>().get_width(); // 480
        const int h = depth_frames.as<rs2::video_frame>().get_height(); //640

        // Create OpenCV matrix of size (w,h) from the colorized depth data
        Mat image(Size(w, h), CV_8UC1, (void*)depth_frames.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);
        waitKey(1);
        }

    return EXIT_SUCCESS;
}

catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

Results : 

[image]

Is there any quick fix for that ??

zivsha, 2018-03-16:
Hi @kmonachopoulos ,
You are using the wrong format for the matrix (`CV_8UC1`), it should be `CV_8UC3`.
Please have a look at the [opencv examples](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv) for more info

kmonachopoulos, 2018-03-16:
Hello @zivsha I have already tried that and get : Segmentation fault (core dumped)

zivsha, 2018-03-16:
Try running the `imshow` example and see if it works. If you still get a seg fault add some more details on it , like a stack trace etc..

ev-mp, 2018-03-19:
Hi @kmonachopoulos ,
The depth frame native format Z16 stands for [16bit/unsigned/little endian](https://01.org/linuxgraphics/gfx-docs/drm/media/uapi/v4l/pixfmt-z16.html).
To render it with grayscale levels you need to convert it to a similar type OpenCV matrix type, probably [CV_16UC1](http://ninghang.blogspot.co.il/2012/11/list-of-mat-type-in-opencv.html)

In the demos the depth frame is often converted into RGB space to enhance visualization traits:  
 16bit grayscale->24bit RGB (8bit per channel).   
In these cases the `CV_8UC3` would be the appropriate conversion type, like in the [following example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/imshow/rs-imshow.cpp#L24).

kmonachopoulos, 2018-03-19:
Thank you @ev-mp ! CV_16UC1 solved the issue !

kmonachopoulos, 2018-03-19:
Solved .. Closing ..

closed 2018-03-19 by kmonachopoulos