realsenseai/librealsense #1382: No device connected

MathiasDePaepe, 2018-03-17:
I'm using pyrealsense2 on the Intel Ready To Fly Drone and when I'm executing a script like in the link   (https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python), I got the error that there is no device connected. But when execute the command cpp-headless, it can output 4 images and it says that there is 1 connected RealSense device, namely Intel RealSense R200. Does someone have an idea what could be the problem?

Thanks in advance

zivsha, 2018-03-17:
Are you using [pyrealsense 2.0](https://pypi.python.org/pypi/pyrealsense/2.0) or [pyrealsense2](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python)?

MathiasDePaepe, 2018-03-17:
Hello,

I installed them both buth for this example I imported pyrealsense2 like in the example.

Thanks

zivsha, 2018-03-17:
The python module `pyrealsense2` uses the `realsense2` library which is aimed at version 2.x.x of librealsense. cpp-headless is an example for librealsense 1.x.x. 
Version 2.x.x fully supports only SR300 and D400 series, R200 is not fully supported and is maintained in the legacy branch only (v1.x.x), which does not work with pyrealsense2.
If cpp-headless works then the library is installed properly and the camera is functioning, so you can try to use pyrealsense 2.0 (which is a community extension).

MathiasDePaepe, 2018-03-17:
Okey thanks for your help, I will try it that way.

closed 2018-03-17 by MathiasDePaepe