realsenseai/librealsense #1383: Masking for pointcloud computation

bloyl, 2018-03-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2                          | 

### Issue Description

I am trying generate pointclouds but only of some objects in the frame. Is there a way to mask the depthframe before handing it off to pointcloud processing block?

I'm using opencv for object detection so I have masks as opencv Mats which I would like to use.
If nothing exists does any one have any suggestions on the must straight forward way to do this?

Thanks.
-Luke

zivsha, 2018-03-17:
Why not apply the maskes Mats on the generated points?

bloyl, 2018-03-18:
My thought is that applying 2D masks to the raw 2D frames might be more
common task that has already been implemented.

Its also not immediately clear how I would get the 2d depth frame indexes
that were used to generate a specific point in the point cloud that would
allow me to remove it. If you have an idea how to do that I'm open to it.

jjoachim, 2018-03-18:
`
rs2::video_frame frame = get_frame(); //however this is done in your code.
int cv_type = (is_dept_frame) ? CV_16UC1 : CV_8UC3;
cv::Mat matt(frame.get_height(), frame.get_width(), cv_type, (uint8_t*)frame.get_data());
`

bloyl, 2018-03-23:
This worked like a charm Thanks!

closed 2018-03-23 by bloyl