realsenseai/librealsense #1384: Why is librealsense version 2.10.2 not tagged?

UnaNancyOwen, 2018-03-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| SDK Version                            |  2.10.2                                                    | 

### Issue Description
Why is librealsense v2.10.2 not tagged?
The version number has already been incremented in #1313.
But, It has not been [tagged](https://github.com/IntelRealSense/librealsense/tags) and [released](https://github.com/IntelRealSense/librealsense/releases) yet.
We still have to wait for it?

dorodnic, 2018-03-18:
Hi @UnaNancyOwen 
We are currently working on better support for USB2 and have merged some patches to allow it, preparing to release it officially as 2.10.2. There are, however, still changes being made on how the device is going to behave when connected to USB2 port, and we are still working to accommodate these changes.
We do not necessarily tie new USB2 support with 2.10.2 release - we plan to make official release this week regardless, with all the ongoing bug-fixes and additions.

UnaNancyOwen, 2018-03-18:
> We are currently working on better support for USB2 and have merged some patches to allow it, preparing to release it officially as 2.10.2. There are, however, still changes being made on how the [...]

It is great to support USB2! 👍
I will look forward to release latest versions.
Thanks,

closed 2018-03-18 by UnaNancyOwen