realsenseai/librealsense #1389: Setting-up Python Wrapper

idang12, 2018-03-19:
I am using the d400 model, firmware v 05.09.02.00.
I am using win10 on a pc and downloaded sdk 2.0 version.
I am having trouble getting started with the Python wrapper.
When I open the 'cmake-gui' I can't get the 'BUILD_PYTHON_BINDINGS' option and I'm not sure how to get it. 
I am reffering to the Python wrapper guide that was published here:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#installation

Help would be appreciated.

zivsha, 2018-03-19:
You need to click "configure" then look under one of the drop downs that are created and you will find this option. Check it, and then click generate

idang12, 2018-03-20:
@zivsha first of all thanks for the response!
If I click "configure" it only gives me options for a generator (Visual Studio from various versions and other generators) but the "BUILD_PYTHON_BINDINGS" is not one of them..

lramati, 2018-03-26:
@idang12 When you click configure for the first time on a project, CMake asks how you intend to build the project. After you select your preferred compiler/IDE CMake will do the initial configuration, at which point the BUILD_PYTHON_BINDINGS option will appear (potentially in the BUILD dropdown group).

idang12, 2018-03-29:
Which compiler do I need for python? 
attached is a screenshot of the error
[image]

lramati, 2018-04-03:
You need a compiler that supports C++11. Visual Studio 2008 does not do this.
As mentioned in the [installation docs](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md), you need Visual Studio 2015 or later.
You can download Visual Studio 2017 from Microsoft's website [here](https://www.visualstudio.com/downloads/).
After making sure Visual Studio 2015 or later is installed on your computer, delete the CMake cache through the option in the File dropdown, and then configure the project again, selecting the appropriate version of Visual Studio as the Generator.

RealSense-Customer-Engineering, 2018-05-10:
[Realsense Customer Engineering Team Comment]
Hi @idang12,
If no other issue, will close this first.

idang12, 2018-05-13:
That is it for now. Thanks for the help!

closed 2018-05-13 by idang12