realsenseai/librealsense #1390: Use long PGP ID for debian install instructions

nh2, 2018-03-19:
On https://github.com/IntelRealSense/librealsense/blame/57e623cd7d5f3912ffae5af86e104d93577fedb0/doc/distribution_linux.md#L17 it recommends

```
sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
```

short PGP keys are trivial to forge (32-bit integers), so it would be better if you could use the long PGP ID instead in the instrall instructions.

dorodnic, 2018-03-20:
Hi @nh2 
Thank you for the suggestion. We will check how our current infrastructure would allow this.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
hi @nh2,
thanks for input, PR created #1978 for this.
let me close this ticket here and please feel free to open new ticket if any question.

closed 2018-07-16 by RealSense-Customer-Engineering