realsenseai/librealsense #1392: load_json() and viewer cannot not work in mac

uncleyoung, 2018-03-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.06.00 | 
| Operating System & Version |    MacOS  | 
| Kernel Version (Linux Only)    | --                                      | 
| Platform                                 | High Sierra 10.13.3  |
| SDK Version                            | SDK 2.0 (Build 2.10.1)    | 

### Issue Description
1. dev.load_json(str) failed when coding like below. the same code works well in ubuntu(with json-file path being modified.)
```
    pipe.start();
    auto profile = pipe.get_active_profile();
    std::ifstream t("/Users/uncle/Documents/Algorithm/IntelRealSense/higacc.json");
    std::string str((std::istreambuf_iterator<char>(t)), std::istreambuf_iterator<char>());
    rs400::advanced_mode dev = profile.get_device();
    dev.load_json(str);
```
here is the error log:
```
RealSense error calling rs2_load_json(dev:0x1018acdc0, json_content:0x102043200, content_size:2986):
    get_xu(id=2) failed! Last Error: No such file or directory
Program ended with exit code: 1
```

2. I cannot tune any params (and other presets) with viewer, it reports an error.
[image]

3. by the way, when sample code running, it always reports `unsupported descriptor subtype: 13`( eg. capture example)

OpenJeDi, 2018-03-21:
I have the exact same issues (also on MacOS 10.13.3) using a D435.

Note that I recorded to a rosbag file on another computer (using windows) and when I load that file instead of using a device, those errors do not occur. I am able to retrieve camera intrinsics from the recording for example, but not from the device. The same code works fine on windows.

dorodnic, 2018-03-21:
We are investigating this problem and will update soon

OpenJeDi, 2018-04-05:
I will as soon as I get the chance (have to deal with some nasty deadlines first).
Thanks a lot for the hard work!

OpenJeDi, 2018-04-06:
I tried a couple of things after building from the development branch (that is the correct one, right?). I tried both with the default cmake settings, and with ENABLE_ZERO_COPY enabled. Both with the same results.

The rs-align example still fails with
`RealSense error calling ():
    Frame of requested stream type was not found!`

The `unsupported descriptor subtype: 13` messages seem to be gone.

When starting realsense-viewer, the following messages are in the console:
`06/04 12:37:31,037 WARNING [0x70000c1f5000] (sensor.cpp:315) Unregistered Media formats : [ 2 P UYVY ]; Supported: [ ]
 06/04 12:37:31,051 WARNING [0x70000c1f5000] (sensor.cpp:315) Unregistered Media formats : [ RW16 ]; Supported: [ ]`

When I try to switch to a different preset, I get this error:
`06/04 12:40:02,572 WARNING [0x7fff97bda340] (types.cpp:57) get_xu(ctrl=1) failed! Last Error: Operation timed out`

When I pull out the USB cable (while streaming), the realsense viewer application hangs.

In my own application, except for the unsupported descriptor message being gone, not much seems to have changed.

This is the error I get when trying to check and set advanced mode (part of it is my own logging):
`[RealSenseCamera::checkAdvancedMode] RealSense exception calling rs2_is_enabled(dev:0x7fa28378ec90, enabled:0):
    get_xu(ctrl=1) failed! Last Error: No such file or directory`

The same when I try to fetch the field of view:
`[RealSenseCamera] Error getting horizontal FOV: set_xu(ctrl=1) failed! Last Error: No such file or directory`

When I force my application to get to the preset json loading part, I get the same set_xu error.

Kind regards,
Jeroen

OpenJeDi, 2018-04-06:
Also, an unrelated problem that still exists (just putting it here): when you don't put in the USB cable quickly enough, the device is recognized as a USB2 device. You have to pull it back out and put the cable back in (quickly and firmly enough) for it to be recognized as a USB3 device and enable all functionality.

baptiste-mnh, 2018-04-11:
Hi,
Is there a way for the moment to set for exemple 
 - param-leftrightthreshold
 - param-regionshrinku
 - param-regionshrinkv

with another way than the loadJson() ?

Maybe the same way than this ? : 
```

//Set preset
    auto sensors = dev.query_sensors();
    rs2::sensor depth_sensor = sensors.at(0);
    rs2::sensor rgb_sensor = sensors.at(1);

    if (depth_sensor.supports(RS2_OPTION_LASER_POWER))
    {
        // Query min and max values:
        auto range = depth_sensor.get_option_range(RS2_OPTION_LASER_POWER);
        depth_sensor.set_option(RS2_OPTION_LASER_POWER, range.max); // Set max power
    }
    if (depth_sensor.supports(RS2_OPTION_GAIN))
    {

        depth_sensor.set_option(RS2_OPTION_GAIN, 30.f);
    }
    if (rgb_sensor.supports(RS2_OPTION_GAIN))
    {
        rgb_sensor.set_option(RS2_OPTION_GAIN,65.f);
    }
```

holstenholsten, 2018-04-12:
Same issue here. Using the latest from master, D415, macOS.

dorodnic, 2018-04-15:
We are still investigating the error. @EladZucker feel free to communicate any of your findings so far.

dorodnic, 2018-04-18:
Should be resolved in [v2.10.4](https://github.com/IntelRealSense/librealsense/releases/tag/v2.10.4), please let us know if you still see the problem

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
hi @uncleyoung, @OpenJeDi, @holstenholsten,

this issue should be fixed already, could you please confirm and let me know if you need more support on this topic.
thanks.