realsenseai/librealsense #1393: I have a problem in csharp library

HungLearner, 2018-03-20:
In d415 tools, i can use "Holes Fill" to fix my depth image, but i can't change this value on csharp library, why?

This is my code:
profile.Device.Sensors[0].Options[Option.HolesFill].Supported

So, i can get "False" output, it doesn't support?

dorodnic, 2018-03-20:
Hi @HungLearner 
It's a bit more complicated then that.
Please fetch the latest SDK from source and build the C# project under `wrappers` directory (just to make sure we are working with the same version).
Then create `SpatialFilter` object like this:
```cpp
filter = new SpatialFilter();
filter.Options[Option.HolesFill].Value = 5.0f;
```
If you are working with `capture` example, modify the main loop by adding a call to `ApplyFilter`:
```cpp
var frames = pipeline.WaitForFrames();
var filtered_depth = filter.ApplyFilter(frames.DepthFrame);
var colorized_depth = colorizer.Colorize(filtered_depth);
filtered_depth.Dispose();
UploadImage(imgDepth, colorized_depth);
```
Then you should see the effect of hole filling. In this example, I have set filter intensity to max, just to demonstrate, but you can play with the `Value` to get better effect for you. 
Of course, the built-in hole-filling algorithm is not the best you can do, but rather something optimized for fast processing.

HungLearner, 2018-03-20:
oh....
That's true!!!
Thanks you!

dorodnic, 2018-03-20:
Also, I would recommend to play with `Visual Preset` option on depth sensor. Built-in visual presets invoke hardware hole-filling when you prioritise dense depth.

HungLearner, 2018-03-20:
OK, I will try to do. Thanks again~

closed 2018-03-22 by HungLearner