realsenseai/librealsense #1394: Unity 2017.3.1f1 (64 bit) crashes every single time when displaying colorised pointcloud data

ashwinjaishanker, 2018-03-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.0-development  | 

### Issue Description
Crash occurs around 30 seconds after running the game. Occurs only when displaying pointcloud data with colorised shaders.

dorodnic, 2018-03-22:
Hi @ashwinjaishanker 
I have reproduced this issue and we are looking into it. From first glance it seems like a very rapid allocation and deallocation of native objects that is causing memory leak, and eventually a crash. 
Will try to resolve ASAP, but in the meanwhile you should not use the colorized point-cloud feature.

infamousdutch, 2018-04-13:
Is there any update on this? We are having the same issue.  Below is the stack trace from the Unity log file:
========== OUTPUTING STACK TRACE ==================

0x00007FFBE3188EC2 (realsense2) rs2_query_sensors
0x00007FFBE31873D2 (realsense2) rs2_query_sensors
0x00007FFBE3188C73 (realsense2) rs2_query_sensors
0x00007FFBE318C590 (realsense2) rs2_query_sensors
0x00007FFBE3155C2D (realsense2) rs2_query_sensors
0x000000003CDA602F (Mono JIT Code) (wrapper managed-to-native) Intel.RealSense.NativeMethods:rs2_process_frame (intptr,intptr,object&)
0x000000003CDA5AF6 (Mono JIT Code) Intel.RealSense.PointCloud:Calculate (Intel.RealSense.Frame)
0x000000003CDA1B57 (Mono JIT Code) [C:\Users\epilnock\Documents\librealsense\wrappers\unity dev\Assets\RealSenseSDK2.0\Scripts\PointCloudGenerator.cs:52] PointCloudGenerator:OnFrames (Intel.RealSense.FrameSet) 
0x000000003CDA173B (Mono JIT Code) [C:\Users\epilnock\Documents\librealsense\wrappers\unity dev\Assets\RealSenseSDK2.0\Scripts\RealSenseDevice.cs:151] RealSenseDevice:HandleFrameSet (Intel.RealSense.FrameSet) 
0x000000003CDA0F4F (Mono JIT Code) [C:\Users\epilnock\Documents\librealsense\wrappers\unity dev\Assets\RealSenseSDK2.0\Scripts\RealSenseDevice.cs:163] RealSenseDevice:ProcessFrames (Intel.RealSense.FrameSet) 
0x000000003AD34B96 (Mono JIT Code) [C:\Users\epilnock\Documents\librealsense\wrappers\unity dev\Assets\RealSenseSDK2.0\Scripts\RealSenseDevice.cs:200] RealSenseDevice:Worker_DoWork (object,System.ComponentModel.DoWorkEventArgs) 
0x000000003AD342CF (Mono JIT Code) System.ComponentModel.BackgroundWorker:OnDoWork (System.ComponentModel.DoWorkEventArgs)
0x000000003AD3390C (Mono JIT Code) System.ComponentModel.BackgroundWorker:ProcessWorker (object,System.ComponentModel.AsyncOperation,System.Threading.SendOrPostCallback)
0x000000003AD33551 (Mono JIT Code) (wrapper runtime-invoke) <Module>:runtime_invoke_void__this___object_object_object (object,intptr,intptr,intptr)
0x00007FFBFB1064AF (mono) [c:\buildslave\mono\build\mono\mini\mini.c:4937] mono_jit_runtime_invoke 
0x00007FFBFB058A19 (mono) [c:\buildslave\mono\build\mono\metadata\object.c:2623] mono_runtime_invoke 
0x00007FFBFB05EF27 (mono) [c:\buildslave\mono\build\mono\metadata\object.c:3838] mono_runtime_invoke_array 
0x00007FFBFB05F1FC (mono) [c:\buildslave\mono\build\mono\metadata\object.c:5490] mono_message_invoke 
0x00007FFBFB07F873 (mono) [c:\buildslave\mono\build\mono\metadata\threadpool.c:1024] mono_async_invoke 
0x00007FFBFB0802CA (mono) [c:\buildslave\mono\build\mono\metadata\threadpool.c:1460] async_invoke_thread 
0x00007FFBFB084017 (mono) [c:\buildslave\mono\build\mono\metadata\threads.c:706] start_wrapper 
0x00007FFBFB139EAB (mono) [c:\buildslave\mono\build\libgc\win32_threads.c:574] thread_start 
0x00007FFC2F111FE4 (KERNEL32) BaseThreadInitThunk
0x00007FFC2F5DEFC1 (ntdll) RtlUserThreadStart

========== END OF STACKTRACE ===========

ashwinjaishanker, 2018-04-19:
Hi @dorodnic,

Is there any update regarding this issue? An approximate timeline? I'm using the fix outlined [here](https://github.com/IntelRealSense/librealsense/issues/1477) for now.

dorodnic, 2018-04-26:
Hi @ashwinjaishanker 
I apologise but this feature is not yet ready for production. I tried the w/a you mentioned and some other approaches but I think that using particle system for point-cloud was an architectural mistake. We discuss rolling-back and trying again with a mesh updated through a shader.

ashwinjaishanker, 2018-05-06:
Thank you for the update, @dorodnic 

A mesh updated through a shader would be very interesting. Look forward to it. Thanks.

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
@ashwinjaishanker
The issue is fixed and merged into development branch, the point cloud is now displayed in dedicated scene. Please have a test and let me know the feedback, thank you.

RealSense-Customer-Engineering, 2018-06-12:
[Realsense Customer Engineering Team Comment]
@ashwinjaishanker
Any update? if not I will close this ticket.

closed 2018-06-14 by RealSense-Customer-Engineering