realsenseai/librealsense #1395: Could CameraOption have a 'ReadOnly' property?

jb455, 2018-03-20:
In my app I change the DepthUnits value for the D400 cameras. I do so like this (C#):
```
if (depthSensor.Options[Option.DepthUnits].Supported)
    depthSensor.Options[Option.DepthUnits].Value = 0.0001F;
```
This is fine for the D400s but I also need to support SR300s. The SR300 returns `true` for `Supported` but throws an exception when I try to change the value saying it's read only. 
It would be good if there was a `ReadOnly` property I can check before attempting to change the value so I don't have to hardcode a device name check (which may change in future) or just try to change the value anyway and suppress the resulting exception.

Thanks!

dorodnic, 2018-03-20:
Hi @jb455 
This is a good point, we will try to offer something like this.

zivsha, 2018-03-20:
Hi,
This already exists starting [3146a739]...(https://github.com/IntelRealSense/librealsense/commit/3146a739c701d78c4a10f1589f4ef2219c13fa95)

jb455, 2018-03-20:
@zivsha Ok, thanks. Is there a reason the latest SDK release doesn't include the C# wrapper pre-built any more? I realised that the dll I'd copied from bin\x64 was from a previous release (the file was dated 11th Feb) as the new installer didn't replace the file. I'll clone from source and build myself for now.

dorodnic, 2018-03-20:
Hi @jb455 @zivsha 
Sorry for misleading, I was not aware of it :)
As for the pre-built DLL, I think it was renamed to `Intel.Realsense.dll`

jb455, 2018-03-20:
Ah right, I hadn't seen the name change mentioned in the release notes or anything. It's working now, thanks!

closed 2018-03-20 by jb455