realsenseai/librealsense #1396: Intel RealSense R200: how to get depth data

MathiasDePaepe, 2018-03-20:
Hello,

I'm new in Intel RealSense and I'm using the R200. Does anyone knows how to get depth data from the frames. I would use these data to determine the distance to an object?

Thanks!

dorodnic, 2018-03-20:
Hi @MathiasDePaepe 
Please take a look at [c-tutorial-1-depth.c](https://github.com/IntelRealSense/librealsense/blob/legacy/examples/c-tutorial-1-depth.c) on the legacy branch.

MathiasDePaepe, 2018-03-20:
Thanks for the fast response, I want to use the following example but the code is not mentioned. https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/dnn

dorodnic, 2018-03-21:
This example was never provided for the R200, but it should be relatively easy to adjust example code to work with librealsense1.

dorodnic, 2018-03-25:
Closing this issue, feel free to open a new one if you have follow-up questions

closed 2018-03-25 by dorodnic