realsenseai/librealsense #1399: Python: the specified module could not be found

FirefoxMetzger, 2018-03-20:
I am using the latest version of the __RealSense SDK 2__ (compiled from source on Windows 10 using Visual Studio Community 2017). In addition to the SDK I am building the python wrappers for __python2.7 (32-bit)__ and __python3.6 (64-bit)__.

Moving the compiled and tested libraries (working on the original machine) to a clean Windows 10 machine (merely the two python versions are installed) I encounter above error.

Interestingly, I can fix this issue for both python versions by installing Visual Studio on said machine. I've tried using the VS 2015 redistributable and VS2017 redistributable instead, however they do not provide the missing libraries.

Which dependency am I missing?

dorodnic, 2018-03-21:
Hi @FirefoxMetzger 
Does this reproduce with precompiled `pyrealsense2.pyd` we provide for **python2.7** inside our installer? 
You can see what library is missing using the [dependencywalker](http://www.dependencywalker.com/). 
In general, assuming you compile your code as **Release**, VS redistributable should be enough. If you compile **Debug**, additional libraries will be needed.

FirefoxMetzger, 2018-03-21:
Ah, yes it was the fact that I was compiling in Debug mode rather then Release mode. That fixed it for me. Thanks!

closed 2018-03-21 by FirefoxMetzger

akapikumin, 2018-06-11:
I am using Windows10 and Python3.6.
I tried to build Python wrapper according to documentation. I finished step5 and tried `import pyrealsense2`.
However, "the specified module could not be found" error was generated as the result.
I compiled code as Release. What else should I try?