realsenseai/librealsense #1410: [.NET] Accessing temperature option

jangernert, 2018-03-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.1 & developement branch (27cc851eec82945171190c2156701f8d4bc29b65) | 

### Issue Description
I'm not sure if this is intended behavior or a bug. Using a pipeline object to handle streams and sensors one does not have to fiddle with sensors themselves, or so I thought.
To get the `DepthScale` form the depth sensor I have to first "open" the sensor or else I just get a `DepthSacale` of 0.
For the `AsicTemperature` and `ProjectorTemperature` I have to call "start" in addition to "open" to get a valid option object.

Here is a minimal sample of the workaround I'm currently using:

```
using (var ctx = new Context())
using (var pipeline = new Pipeline(ctx))
using (var config = new Config())
{
	config.EnableStream(Stream.Depth, -1, 640, 480, Format.Z16, 30);
	pipeline.Start(config);
	FrameSet frames = pipeline.WaitForFrames();
	var dev = ctx.Devices[0];
	var depthSensor = GetSensor(dev, SensorNames.Stereo);
	depthSensor.Open();
	FrameQueue q = new FrameQueue();
	float depthScale = GetSensor(dev, SensorNames.Stereo).DepthScale;
	depthSensor.Start(q);
	bool supported = GetSensor(dev, SensorNames.Stereo).Options[Option.AsicTemperature].Supported;
	depthSensor.Stop();
	depthSensor.Close();
	pipeline.Stop();
}
```

`GetSensor` just iterates over `Sensors` of the device and returns the one with the desired name.
If I don't open and start the sensor manually before getting the `AsicTemperature` option, supported always returns `false`, the value throws and exception and max and min are just 0.
Is it supposed to work like this, or am I missing something?

dorodnic, 2018-03-21:
At the moment, we only support querying temperature during streaming, and not in idle mode.

jangernert, 2018-03-21:
So a running pipeline is not considered "during streaming"? I am not familiar with the internals of librealsense, so this seems a bit confusing to me.

dorodnic, 2018-03-21:
Hi @jangernert 
I got somewhat confused by the code-snippet.

```cs
using (var pipeline = new Pipeline())
using (var config = new Config())
{
    config.EnableStream(Stream.Depth, -1, 640, 480, Format.Z16, 30);
    var profile = pipeline.Start(config);
    var dev = profile.Device;
    FrameSet frames = pipeline.WaitForFrames();
    bool supported = dev.Sensors.First().Options[Option.AsicTemperature].Supported;
    var scale = dev.Sensors.First().DepthScale;
    pipeline.Stop();
}
```
This code should offer the same results, but without having to manually open the sensor.

jangernert, 2018-03-21:
Ah, this seems to work.
So having a different handle to the same physical depth sensor is the problem in this case?

dorodnic, 2018-03-21:
Yes, we do not cache sensor objects in the library, every time you get list of sensors you create new set of Media-Foundation handles and state. Ideally it would make sense to cache, but we are concerned about 
threading in connect-disconnect flows with shared mutable state.

jangernert, 2018-03-21:
Okay, thx for the insight.
Hopefully this will help others googling for this problem as well.

closed 2018-03-21 by jangernert