realsenseai/librealsense #1414: How can I retrieve the distance or depth data with Intel RealSense D435 depth camera?

StevenWong1995, 2018-03-22:
Hi everyone, does anyone know how can I retrieve the distance of an object or a person from the sensor of the RealSense D435 depth camera?

Is there any specific function call or algorithm that I should use to do that? I'm using C# to do this. 
My idea is to use face detection from OpenCV to integrate with the depth sensor to get the distance of a person standing away from the camera. 

Thanks in advance.

dorodnic, 2018-03-22:
Hi @StevenWong1995 
Please let us know if this helps: [wrappers/opencv/dnn](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/dnn)

StevenWong1995, 2018-03-22:
@dorodnic Hi! Thanks for the reference there. This definitely helps but Deep Neural Network is kind of difficult for me to achieve as for my level. By the way, I've managed to obtain the distance by capturing the frame data and process them into a simpler face tracking algorithm using OpenCV as well. 

Still thank you very much! 👍

dorodnic, 2018-03-22:
If you ignore all of the DNN stuff, it can still help you get mean depth from a rectangular region of interest. 
Please let us know if you have more follow-up questions

closed 2018-11-26 by dorodnic