realsenseai/librealsense #1415: no pyrealsense2.pyd after building pyrealsense2 project

Jowizo, 2018-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.08.15.00) | 
| Operating System & Version |   {Win (10  |
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.10.1 }                          | 

I'm having issues with the python wrapper. 
To start off following the setup instructions I ran CMake with the following parameters:
[image]

Then I opened the librealsense.snl file that was created in the ..\librealsense-master\build folder. Once open in visual studio I 'built solution' and it build without any errors. Here's a screenshot of the configuration manager: (the only two builds that were unchecked are install and uninstall)

[image]

According to the instructions there would be a directory build\x64-Release\Release\  containing the pyd file, which I don't see.

Any help would be highly appreciated!

Jowizo, 2018-03-22:
Thanks for your reply @dorodnic, I did that and found a file named pyrealsense2.cp36-win_amd64.pyd! For some reason I tried searching for the realsense2.pyd specific file. I'm still having some trouble importing the package in python after both adding the .pyd file to the site-packages and adding both the .pyd and .dll files next to my script. Would I have to rename the pyd file to pyrealsense2.pyd? 

Here's a screenshot of the build directory:
[image]

dorodnic, 2018-03-22:
I think renaming it would help, but I'm not 100% sure why it is generated this way.
We also provide precompiled `pyrealsense2.pyd` in the installer (just as another fallback if you encounter problems)

Jowizo, 2018-03-23:
Where is this file located?

dorodnic, 2018-03-24:
`C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64\pyrealsense2.pyd` I believe.

dorodnic, 2018-03-27:
This can happen if you are using python3. At the moment we only pre-compile for python2.7 (adding pre-compiled version for python3 is in our backlog)

RealSense-Customer-Engineering, 2018-03-28:
[Realsense Customer Engineering Team Comment] 
@Jowizo, please just rename the pyrealsense2.cp36-win_amd64.pyd to pyrealsense2.pyd, and then copy pyrealsense2.pyd and realsense2.dll to your Python folder (yours should be C:\Programe Files\Python36-64\Lib\site-packages.

After that, just open the command line and type python to get into python environment, then type :
>>> import pyrealsens2 as py

if there is no error, which mean the environment is set correctly, you can then run the example.

###

Jowizo, 2018-03-28:
Thanks for the help guys, it was a configuration mistake on my end. Working like a charm now!

RealSense-Customer-Engineering, 2018-03-28:
[Realsense Customer Engineering Team Comment] 
great to hear that, close it.

closed 2018-03-28 by RealSense-Customer-Engineering