realsenseai/librealsense #1417: Load .json when stream from .bag file

vladsterz, 2018-03-22:
Required Info |  
-- | --
Camera Model | D415 and D435
Operating System & Version | Win 10
Platform | PC
SDK Version | 2.10.1

### Issue Description
How to load a .json file when I am playing back from a .bag file?

dorodnic, 2018-03-22:
Hi @vladsterz 
You can't load JSON when running from recording. All depth-parameters has been recorded during the live run and you can't modify them retroactively (JSON parameters actually configure the hardware, not just software post-processing).
I hope I understood correctly what you are trying to do.

vladsterz, 2018-03-22:
Don't JSON file save as well post-processing parameters? I want to load some post processing parameters to recorded .bag files. Is there anyway to achieve it?

dorodnic, 2018-03-22:
I see. At the moment, post-processing settings are not saved into the JSON file. When we do add this capability, you are right that we will also need to allow loading relevant parts of the JSON when running playback.

vladsterz, 2018-03-22:
Ok, thanks alot.

closed 2018-03-22 by vladsterz