realsenseai/librealsense #1418: Question of firmware with D415 camera

sbnn123456, 2018-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (latest) | 
| Operating System & Version |   {Win (10) }| 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {  }                          | 

### Issue Description
It there any schedule  of the newest firmware
which can set the  D415 camera in slave mode to sync multi-camera?

any help will be appreciated.

RealSense-Customer-Engineering, 2018-04-03:
[Realsense Customer Engineering Team Comment] 
@sbnn123456, there is not a scheduled date for the slave mode f/w yet. Currently validation testing is being performed to ensure reliable software/hardware operation. You may monitor the Intel RealSense pages for firmware updates or periodically check back in here.

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
@sbnn123456, as previously noted, we don't have a specific schedule set so I would suggest that you monitor the Intel RealSense pages for updated firmware or check back in with a new Github issue in a month or so; for the time being I will close this one.

closed 2018-04-17 by RealSense-Customer-Engineering